FreeCAD/FreeCAD #15392: AttachEngine3D: subshape not found, but i don't used subshape bindes (and recompute failed)

amreo, 2024-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The sketch003 in joint-top-case.fcstd in this zip [joint-case.zip](https://github.com/user-attachments/files/16196212/joint-case.zip) shows the error "AttachEngine3D: subshape not found" and the recompute fails, and I don't undestand what I done wrong.

Steps to reproduce:
1) unzip the two *.fcstd files [joint-case.zip](https://github.com/user-attachments/files/16196212/joint-case.zip)
2) open joint-top-case.fcstd
3) hover the mouse of Sketch003 on tree view
4) view the error
5) Press ctlr+R to recompute
6) In the report view should show the error "Recompute failed!"

I think the bug impacts the UX because makes the error unclear to the user

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * fasteners 0.5.0
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * ModernUI (Disabled)
  * Beltrami 1.0.8-alpha
  * CurvedShapes 1.0.5
  * boltsfc 2022.11.5
  * ThreadProfile 1.89.0
  * sheetmetal 0.4.0
  * lattice2 1.0.0
  * Curves 0.6.22
  * Assembly4 0.50.6
  * Assembly3 0.12.2
  * A2plus 0.4.63
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-13:
could you try it in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

amreo, 2024-07-13:
I tested on the dev version:
```
OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git) AppImage
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
```

The error seems to be the same: "AttachEngine3D: null subshape". I do not seem `recompute failed` error.
At least I found out that it is underconstrained by two DoF and 
```
08:14:24  joint-top-case#Sketch003: Shape index 92 out of bound 76
08:14:24  joint-top-case#Sketch003: Shape index 87 out of bound 76
```

maxwxyz, 2024-07-13:
@bgbsww is this related to tnp-fixes?

luzpaz, 2024-08-25:
@bgbsww :point_up_2:

bgbsww, 2024-08-25:
Title says a 0.21.2 bug.  Thus cannot be TNP.  On opening, you can see that there are external references in Sketch003 that are to elements in Pad001 (Face36) way larger than the count of geometry in Sketch001, which I can't see how it would work.

Did maybe Sketch001 get edited to remove geometry after the external constraints were set up in Sketch003?  And if so, under which version of the code that failed to detect this, and resulted in a broken file?

amreo, 2024-12-12:
I reproduced it with freecad 1.0 
```

amreo, 2024-12-12:
I reproduced it with the version 1.0
```
OS: KDE Flatpak runtime (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.25
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * ModernUI (Disabled)
  * Beltrami 1.0.8-alpha
  * CurvedShapes 1.0.9
  * boltsfc 2022.11.5
  * ThreadProfile 1.92.0
  * sheetmetal 0.4.23
  * lattice2 1.0.0
  * Curves 0.6.42
  * Assembly4 0.50.6
  * Assembly3 0.12.2
  * A2plus 0.4.65
```

The bug still applies, but I had to migrate the file version and I don't know if the file is okay or "corrupted".
The report says:
```
23:21:34  <Sketch> SketchObject.cpp(9050): Failed to project external geometry in joint_top_case#Sketch003: Pad001.Edge92
Shape index 92 out of bound 76
23:21:34  <Sketch> SketchObject.cpp(9050): Failed to project external geometry in joint_top_case#Sketch003: Pad001.Edge87
Shape index 87 out of bound 76
23:21:40  Sketch003: AttachEngine3D: subshape not found Pad001.Face36
```

FEA-eng, 2025-10-04:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2026-10-04:
Opening the file i see:
```
11:14:01  <Sketch> SketchObjectExternal.cpp(2627): Failed to project external geometry in joint_top_case#Sketch003: Pad001.Edge92
Shape index 92 out of bound 76
11:14:01  <Sketch> SketchObjectExternal.cpp(2627): Failed to project external geometry in joint_top_case#Sketch003: Pad001.Edge87
Shape index 87 out of bound 76
11:14:01  joint-top-case#Sketch003 SketchObject::onChanged : Unmanaged change of Constraint Property results in invalid constraint indices
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:03  <TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch
11:14:04  PositionBySupport: AttachEngine3D: subshape not found Pad001.Face36
11:14:04  Sketch003: AttachEngine3D: subshape not found Pad001.Face36
```

```
OS: Manjaro Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 26.3.0dev.20261001 (Git shallow) AppImage
Build date: 2026/10/01 12:05:00
Build type: Release
Branch: (HEAD detached at 99c5620c5)
Hash: 99c5620c5f739ca5c308849dda95ac468fec8b79
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/124.55/2
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.2-arch1.1/Intel/Mesa Intel(R) UHD Graphics 620 (WHL GT2)
```