FreeCAD/FreeCAD #15393: WB Arch: create structure -> structure option -> beam without category selected

ssl33, 2024-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

WB Arch: creating Beam. Details in topic:  https://forum.freecad.org/viewtopic.php?style=4&t=77050 (please find link to YT movie )
In current version FC (0.21.2) looks the same problem.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * A2plus 0.4.65
  * Assembly4 0.50.14
  * BIM021 2021.12.0
  * DynamicData 2.60.0
  * ExplodedAssembly
  * fasteners 0.5.15
  * freecad.gears 1.2.0
  * Help 1.0.3
  * OpenDark 2023.12.17
  * parts_library
[/code]
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-13:
@ssl33 what version of latest 0.22 dev has this been reproduced on ? Can  you edit ans paste that into your original post perhaps ?

ssl33, 2024-07-14:
I didn't test in version 0.22 dev, but in the latest stable version 0.21.2.33771.

luzpaz, 2024-07-21:
@yorikvanhavre any chance you can reproduce on dev version ?

ssl33, 2024-07-22:
I just tested on version 0.22 dev (38261 git). It works incorrectly.

yorikvanhavre, 2024-07-30:
I can't reproduce this with current code - can you try a development version?

[image]

However, there is still a bug, beams drawn at an arbitrary angle can get wrongly rotated around their axis. That is because of the rotation method used (from vector/to vector). We need to find a better way that keeps an up vector.

luzpaz, 2024-07-30:
> However, there is still a bug, beams drawn at an arbitrary angle can get wrongly rotated around their axis. That is because of the rotation method used (from vector/to vector). We need to find a better way that keeps an up vector.

@yorik open a new ticket for that

yorikvanhavre, 2024-07-31:
Good idea - done #15704

luzpaz, 2024-07-31:
@ssl33 I'm closing this ticket for now. If you can recreate the bug on latest development revision, please by all means re-open it. TIA

closed 2024-07-31 by luzpaz

ssl33, 2024-08-01:
ok