FreeCAD/FreeCAD #15395: Sketcher: Sketches with non-reference points cause a crash

alexneufeld, 2024-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

- create a new document
- create an empty sketch
- toggle 'construction geometry' to OFF
- create a point in the sketch
- close the sketch
- FreeCAD crashes:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7f57f185b050]
#1  0x7f57f4cfbc5a in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::_M_mbegin() const from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0xc
#2  0x7f57f4cfbb3c in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::_M_begin() from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x18
#3  0x7f57f4d21306 in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::lower_bound(Data::MappedName const&) from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x2c
#4  0x7f57f4d1c713 in std::map<Data::MappedName, Data::IndexedName, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::lower_bound(Data::MappedName const&) from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x23
#5  0x7f57f4d187b4 in std::enable_if<std::is_constructible<std::pair<Data::MappedName const, Data::IndexedName>, std::pair<Data::MappedName, Data::IndexedName> >::value, std::pair<std::_Rb_tree_iterator<std::pair<Data::MappedName const, Data::IndexedName> >, bool> >::type std::map<Data::MappedName, Data::IndexedName, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::insert<std::pair<Data::MappedName, Data::IndexedName> >(std::pair<Data::MappedName, Data::IndexedName>&&) from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x2c
#6  0x7f57f4d0bcf6 in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x262
#7  0x7f57f4d0c842 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x53e
#8  0x7f579d2ab6c9 in Data::ComplexGeoData::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /home/alex/Code/freecad_build/Mod/Sketcher/Sketcher.so+0x61
#9  0x7f579d26f1a1 in Sketcher::SketchObject::buildShape() from /home/alex/Code/freecad_build/Mod/Sketcher/Sketcher.so+0x221
#10  0x7f579d26ecd7 in Sketcher::SketchObject::execute() from /home/alex/Code/freecad_build/Mod/Sketcher/Sketcher.so+0x393
#11  0x7f57f4a29d8f in App::DocumentObject::recompute() from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x75
#12  0x7f57adc90049 in Part::Feature::recompute() from /home/alex/Code/freecad_build/Mod/Part/Part.so+0x1d
#13  0x7f57f495024f in App::Document::_recomputeFeature(App::DocumentObject*) from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x1eb
#14  0x7f57f494db7e in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x986
#15  0x7f57f4ae3c93 in App::DocumentPy::recompute(_object*) from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0x2c7
#16  0x7f57f4adb699 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /home/alex/Code/freecad_build/lib/libFreeCADApp.so+0xf3
#17  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9ba1) [0x7f57f31c9ba1]
#18  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7f57f317be8d]
#19  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7f57f3109c63]
#20  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x3ccd3c) [0x7f57f33ccd3c]
#21  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xc1) [0x7f57f3268891]
#22  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2b7359) [0x7f57f32b7359]
#23  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x6b) [0x7f57f32b850b]
#24  0x7f57f3b0bad1 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/alex/Code/freecad_build/lib/libFreeCADBase.so+0xed
#25  0x7f57f6618a2a in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0xc6
#26  0x7f57f66186e3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0xfd
#27  0x7f57f6619871 in Gui::Command::updateActive() from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x3d
#28  0x7f5714638d74 in SketcherGui::ViewProviderSketch::unsetEdit(int) from /home/alex/Code/freecad_build/Mod/Sketcher/SketcherGui.so+0x2e8
#29  0x7f57f6b43379 in Gui::ViewProvider::finishEditing() from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x31
#30  0x7f57f650b0d2 in Gui::Document::_resetEdit() from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x100
#31  0x7f57f6472028 in Gui::Application::setEditDocument(Gui::Document*) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x94
#32  0x7f57f650afcf in Gui::Document::resetEdit() from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x23
#33  0x7f57f6580e69 in Gui::DocumentPy::resetEdit(_object*) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x4f
#34  0x7f57f657c793 in Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0xf3
#35  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9ba1) [0x7f57f31c9ba1]
#36  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7f57f317be8d]
#37  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7f57f3109c63]
#38  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x3ccd3c) [0x7f57f33ccd3c]
#39  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xc1) [0x7f57f3268891]
#40  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2b7359) [0x7f57f32b7359]
#41  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x6b) [0x7f57f32b850b]
#42  0x7f57f3b0bad1 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/alex/Code/freecad_build/lib/libFreeCADBase.so+0xed
#43  0x7f57f6618a2a in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0xc6
#44  0x7f57f66186e3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0xfd
#45  0x7f57146a1453 in SketcherGui::TaskDlgEditSketch::reject() from /home/alex/Code/freecad_build/Mod/Sketcher/SketcherGui.so+0x19f
#46  0x7f57f6a654eb in Gui::TaskView::TaskView::reject() from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x9d
#47  0x7f57f6a68a5d in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::TaskView::TaskView::*)()>::call(void (Gui::TaskView::TaskView::*)(), Gui::TaskView::TaskView*, void**) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x6b
#48  0x7f57f6a685a1 in void QtPrivate::FunctionPointer<void (Gui::TaskView::TaskView::*)()>::call<QtPrivate::List<>, void>(void (Gui::TaskView::TaskView::*)(), Gui::TaskView::TaskView*, void**) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x42
#49  0x7f57f6a67e45 in QtPrivate::QSlotObject<void (Gui::TaskView::TaskView::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) from /home/alex/Code/freecad_build/lib/libFreeCADGui.so+0x80
#50  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8f4f) [0x7f57f1ce8f4f]
#51  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2fe860) [0x7f57f2afe860]
#52  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8f7c) [0x7f57f1ce8f7c]
[... cut ...]
```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Unknown
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Curves 0.6.39
  * sheetmetal 0.4.18
  * fasteners 0.5.24
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-07-12:
There are six duplicates of #15102 in seven days!

FEA-eng, 2024-07-12:
> There are six duplicates of #15102 in seven days!

Are you sure they all have the same cause? Then maybe the original one should be renamed since it only mentions chamfer currently which can be misleading and cause users to overlook that issue.

wwmayer, 2024-07-12:
They all have this call in the call stack: `Data::ComplexGeoData::setElementName` where the class member to a ElementMap is a null pointer.

closed 2024-07-13 by maxwxyz