FreeCAD/FreeCAD #15399: Views impossible to restore after closing them on docked state

YosypC, 2024-07-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/user-attachments/assets/8ecb89c4-374f-4fcf-a6be-5983626e0af0

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

YosypC, 2024-07-13:
Also:

https://github.com/user-attachments/assets/cda250c9-0a68-4eeb-9f95-e33a7c0c89d1

FEA-eng, 2026-06-13:
Seems still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260609 (Git shallow)
Build date: 2026/06/09 14:33:40
Build type: Release
Branch: grafted,grafted
Hash: 9cb82b95e9f52f3ac2edc9d42525b39a1d77532a
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-09-12:
Still relevant (at least for Report View in overlay mode)

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 616.92/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```