FreeCAD/FreeCAD #15400: macOS: Transparent items not visible (navicube, planes, rotation center)

maxwxyz, 2024-07-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On macOS it seems that all transparent UI objects are not legible or even not recognizable, like the navicube, rotation center indicator, planes, origins,...
[image]
[image]

It is more legible with the dark theme and classic theme but also way worse than on Windows.
Light theme is not usable on macOS like this @kadet1090 @MisterMakerNL FYI

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git)
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-13:
hovering the navicube improves the situation temporarily: 
[image]

With OpenLight the contrast is a bit better but still worse than on Windows:
[image]

Rotation indicator also not visible if you have no object behind it:
[image]

luzpaz, 2024-08-24:
@johnocfii any chance you can test to confirm on macOS ?

JohnOCFII, 2024-08-24:
> @JohnOCFII any chance you can test to confirm on macOS ?

Can confirm.  

[image]

luzpaz, 2024-08-24:
@chennes this is kind of big deal for macOS, should it be a blocker ?

chennes, 2024-08-24:
It makes the light theme difficult to use, but Classic is effectively unaffected. IMO it does not block the release.

maxwxyz, 2024-08-25:
@chennes it's the same for all themes on my system, here is classic:
[image]
I mean still better than light but worse than Windows

chennes, 2024-08-26:
True, but the Navicube labels not showing is not serious enough, in my opinion, to warrant blocking the release. Is there any hint of a possible fix?

MisterMakerNL, 2024-08-26:
@chennes Switching Qstyle probably.

tomverbeure, 2024-11-16:
My 2 cents: I was ready to throw out FreeCAD as unusable on a Mac.

It took me half an hour to figure out that I should have switched from FreeCAD Light to FreeCAD classic, and that's only because I know GitHub well enough to go to the issues, filter on "MacOS" and find this specific issue.

MisterMakerNL, 2024-11-16:
@tomverbeure which Freecad version are you using?

maxwxyz, 2024-11-16:
@hyarion ^ the auto light mode on mac could lead to these issues

tomverbeure, 2024-11-16:
> @tomverbeure which Freecad version are you using?

I just downloaded 1.0RC4. I had no FreeCAD preferences installed. The initial startup screen only shows FreeCAD Light and FreeCAD Dark as theme options, so I choose Light. I didn't even know a FreeCAD Classic theme existed.

MisterMakerNL, 2024-11-16:
@tomverbeure We have only a handful of people here testing on a Mac. So you could really help out if you want.

tomverbeure, 2024-11-16:
Feel free to ping me to try test builds. Not sure if I can help with anything else.

hyarion, 2024-11-16:
This issue is available on macOS when using Light Theme on (System) Light Mode on 38871 too which does not have the auto-theme selection. So it's an old issue.

While this is clearly an issue, I don't think it's a blocker (should be fixed in 1.0.1)

I've also tested on RC4:
<details><summary>macOS + RC4</summary>
<p>

```
OS: macOS 14.5
Architecture: arm64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

</p>
</details> 

Light theme in dark mode:
[image]

Light theme in light mode:
[image]

Maybe changing the opacity to 75% when inactive? (Display > Navigation > Navigation Cube > Opacity when inactive)
Light theme in light mode with that adjusted:
[image]
[image]

BanditoB1, 2024-12-30:
I just noticed the same thing.  Unless you are hovering over the Navigation Cube in MacOS on ARM, the labels for top, bottom, etc. do not show up.

As another Mac user, if you need someone to help test this out, please let me know.

maxwxyz, 2025-05-24:
seems to be fixed in 1.1

closed 2025-05-24 by maxwxyz