FreeCAD/FreeCAD #15402: Empty SubShapeBinder when binding part of a sketch in another body

depthoffocus, 2024-07-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

(I've searched as much as I can and can't find anything immediately matching)

When I create a sub shape binder from only some edges in a sketch that is in the same body, it succeeds, and I can pad that binder.

If I try to create the sub shape binder in another body, with the same selections, it is empty. (Creating a binder from the whole sketch will succeed.)

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git)
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * Curves 0.6.40
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-14:
It does work in the stable 0.21.2 version, so that is a regression.
Can confirm on 
<details><summary>Windows 11, 0.22dev</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.22

``` 

</p>
</details> 

0.21.2 stable works:
[image]

Here is a demo file:
[sub-binder-fail.zip](https://github.com/user-attachments/files/16216571/sub-binder-fail.zip)

@bgbsww could this be related to the TNP fixes?

bgbsww, 2024-08-11:
@maxwxyz Maybe, but also I can't seem to replicate it now if I do the steps in the video.  The attached file needs a recompute on Binder001 to make it start working.   Any chance this is solved?

depthoffocus, 2024-08-12:
Will do, yes. Though I tested with 38429 yesterday and I agree that the problem identified in my video appears to be resolved there.

depthoffocus, 2024-08-13:
@maxwxyz it does indeed work in 38459.

[image]

closed 2024-08-13 by maxwxyz