FreeCAD/FreeCAD #15405: Unhandled Base: Exception caught in GUIApplication:

GWeliver, 2024-07-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Unhandled Base: Exception caught in GUIApplication::notify. Then error message is unknown document '' 
also
11:38:41 QbackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.65
  * Assembly3 0.12.2
  * Assembly4 0.50.13
  * CurvedShapes 1.0.9
  * Curves 0.6.39
  * ExplodedAssembly
  * fasteners 0.5.0
  * Help 1.0.3
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * sheetmetal 0.3.16
[/code]
```

### Subproject(s) affected?

Sketcher

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-07-13:
Is there a procedure to reproduce the issue?

GWeliver, 2024-07-13:
This is a very simple project. This is a reproduction on a 1 trace layer
printed circuit board. Everything will be in layers stacked on the XY plane
only.

Freecad just kept not responding while I was in the sketcher workbench
creating thru hole pads for the various components (resistors,
potentiometer, capacitors, IC, DC power jack, etc.). I have an image of the
circuit board (refer to attachment) and I created 64 of the pads and now I
must start over.

What I originally desired was to rotate my photo slightly. so the bottom of
board (image) would be parallel with the X axis. I calibrated my image.
Then working on XY plane, in part workbench created new sketch (XY) added a
rectangle, little bigger than image, to the positive X, positive Y
quadrant. Then closed sketch. Selected image then ctrl selected sketch and
opened edit and alignment, and did the point1, for image and point 1 for
rectangle (upper left corner for both) then repeated for point 2,3,4
respectively, right clicked and closed.

The bottom edge of image was now parallel as desired. Deleted the sketch
and then started a new sketch. After 64 pads, all constraints and
everything was green, then I went to realign a couple of pads (C) while in
sketcher and it stopped and issued various errors.

I have started over and plan on created sketches with fewer pads then
extrude to 1.6 mm and see if this works?

Side note: In the alignment when you select what needs to be aligned, the
first item you select will be aligned to what is selected second, thus the
first selection will show up on the left, then there is a white vertical
division line and then the second selection is on right side. My image was
on the left, the line and then the rectangle I created in my sketch was on
the right.

There is a possibility I suppose I may have just deleted the rectangle and
used the same sketch and started working from there. Not sure, I usually
never would do that but I have never had this trouble before?

What makes me wonder is the message (QBackingStore::endPaint() called with
active painter; did you forget to destroy it or call QPainter::end() on it?

Thanks for your help

G_Dub

maxwxyz, 2024-07-14:
@GWeliver does this also happen with the current dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Could you provide a demo file?

GWeliver, 2024-07-14:
Yes, I will send the file, however this error may have deleted the sketch
that I was using when it just stopped working.
 I have started over and have not had any problems so far.
FYI: I decided to break the project up into smaller groups, by creating
pads for just resistors, capacitors, headers, DC power switch, etc all on
the same layer. Now starting a new layer for the traces seen from the top,
then there will be traces as seen from the bottom (ground - ). The board
will be 1mm thick, each layer offset -0.10 mm, and a thickness of 0.10mm.

I will send the corrupt file and the also send the new file, when I get
home tonight.

Thanks,
G_Dub

maxwxyz, 2024-09-01:
@GWeliver bump

luzpaz, 2024-09-10:
@GWeliver ping

luzpaz, 2024-09-12:
Closing. If OP responds we can re-open.