FreeCAD/FreeCAD #15408: FEM: Tree objects linked to FEMMeshGmsh are permanently grayed out

FEA-eng, 2024-07-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

0.21.2:

[image]

0.22:

[image]

@marioalexis84 Is it still #14637 after #14771 ?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git)
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-14:
@NomAnor just to confirm: this has nothing to do with the tree visibility changes?

FEA-eng, 2024-07-14:
ElementGeometry2D and ElementGeometry1D are also grayed out. I wonder if #14637 is back.

NomAnor, 2024-07-14:
> @NomAnor just to confirm: this has nothing to do with the tree visibility changes?

I don't think so. This is the code that sets the gray-out:
https://github.com/FreeCAD/FreeCAD/blob/4bb6d191b4686919a2f247bcfce6c186703a3d86/src/Gui/Tree.cpp#L5313-L5335

And this is where the `visible` status is calculated:
https://github.com/FreeCAD/FreeCAD/blob/4bb6d191b4686919a2f247bcfce6c186703a3d86/src/Gui/Tree.cpp#L5261-L5287

Maybe the new python base class from #14771 is missing something or for some reason the above code has a bug.

I would try to figure out what code path in the above code is used (put some logging statements in there) and then figure out where the values that are used in that code path come from on the python side (some prints in the python methods).

PR #15454 (2024-07-21): Fem: Use own base class for Python objects - fixes #15408
This PR fixes #15408. A regression was introduced in https://github.com/FreeCAD/FreeCAD/pull/14793
commit d19b4adb (2024-07-21): Fem: Use own base class for Python objects - fixes #15408
closed 2024-07-21 by chennes