FreeCAD/FreeCAD #15411: ColorPerFace test fails with latest libraries

chennes, 2024-07-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In ColorPerFaceTest there is a test for MultiFuse with the following assert:
https://github.com/chennes/FreeCAD/blob/2e39b4167fd7353e10eb335d751d11a92140d966/src/Mod/Part/parttests/ColorPerFaceTest.py#L159

When I run this with the latest LibPack release candidate 3, I get a failure here. The code is yielding seven faces, rather than the expected eleven (and therefore also colors, etc.). I don't understand why 11 is expected, when the resulting fused object does indeed only have seven faces.

This LibPack is compiled with:
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38121 (Git)
Build type: Release
Branch: main
Hash: d838e4452149717741e22c7dbd7eb2c1c393ce4f
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-07-14:
@wwmayer any idea what's going on here? Is this a change in OCCT's behavior?

wwmayer, 2024-07-14:
Maybe newer OCCT versions perform directly a refinement now where the 3 top and the 3 bottom faces are joined. This would explain why it has 4 faces less.

wwmayer, 2024-07-14:
OK, the problem is something else. If the refinement parameter is enabled then Part::MultiFuse will perform the refinement. So, what's needed is an extra call in testMultiFuse to explicitly switch off the refinement.

commit 45a7d9ca (2024-07-14): Part: Fixes #15411: ColorPerFace test fails
closed 2024-07-14 by chennes