FreeCAD/FreeCAD #15427: Post processor folder missing. Files export as "file" type.

CaptainKokomo, 2024-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I exported a pocket job. completed it.
I create a profile job on the same piece 20 minutes later. I have no post processors in the dropdown menu. I try forcing a file type and it defaults to "gcode for laser engraving" and the spindle will not start up when I try to run any job now.

### Full version info

```shell
Don't know what more to say. AI tells me that I need to look in the "post" folder that is inside my path folder inside the mod folder.... no post folder. AI tells me to go to the github page to download the post folder.... no such folder. I don't know alot about this shit and I am running out of patience very quickly as my cnc has only been back up and running less than a week after having to wait over a month for it to be fixed properly. 
I haven't changed ANYTHING from the first to the second job.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-15:
PLease add the full about info and steps how to reproduce. It would be great if you could share a demo file.

CaptainKokomo, 2024-07-15:
that's the problem I don't know how to reproduce it. Once it happened... I can't undo it to redo it. Which part do you need me to reproduce? I am happy to post anything I can but I 'm not exactly sure what to post... Let me see what I have.
[image]

I have changed the default opening program from Lasergrbl to UGS so thats one issue solved. The spindle still won't spin up and I can't really reproduce the issue as I never really had post processors in that location. I never had a Post folder in the location it's asking for. I honestly don't know what happened between the first job and the second. I'm slowly working it out but the spindle seems to be a sticking point. The cnc mashine is still very new and its got a new control board, (that was a nightmare). I think my issue is that I don't have the post processor in the right spot so I can export as the right file... I don't have any post processors right now.

If you have any suggestions I would greately appreciate the help.... if not... I will try and work it out on my own.
Thank you for reading this public safety message.
Jon

CaptainKokomo, 2024-07-15:
Sorry I dont know what that image was meant to show you. maybe that I don't have post processors in the right location.

luzpaz, 2024-07-18:
@CaptainKokomo ping

CaptainKokomo, 2024-07-20:
Sorry
Standard install of freecad.
Seems the profile job makes UGS think the job is a laser job and it ignores the M3 command. The gantry follows the path but the spindle doesn't spin up. I have fixed it once before but I don't remember how. I left it off and unpluggd for 2 days and went to show my wife what happend and it suddenly was working with a pocket job. Now I tried to create another profile job in freecad and in carveco maker (after the issue had started) to try and trouble shoot to see if it carveco would work but still nothing.

luzpaz, 2024-07-24:
>  Standard install of freecad. 

We ask for full About info for a reason. There are a few ways to install FreeCAD via windows and that also means different 3rd party dependencies that may be the culprit sometimes for bugs. It saves everyone time when this procedure is followed.

closed 2024-08-05 by maxwxyz