FreeCAD/FreeCAD #15430: Core: Selection observer gives bad subnames

PaddleStroke, 2024-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A class that is a selection observer (example TaskAssemblyCreateJoint in jointObject.py) has bad sub_name received :

```
    def addSelection(self, doc_name, obj_name, sub_name, mousePos):
        print(sub_name)
```

Prints :

Body.Pad.;g233;SKT;:H102d,E;FAC;:H102d:4,F;:G0;XTR;:H102d:8,F.Face6
But apparantly only for PartDesign bodies. For others objects (box for example) then it's ok.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-07-15:
@bgbsww that sounds like a topo naming thing no?

bgbsww, 2024-07-15:
Yes, although the question is which answer is correct, which will require trying the methods on LS3 to see.  I suspect missing c++ to python layer code interface changes either way.

PaddleStroke, 2024-07-16:
I have just tested in LS3 and the same happens. So perhaps it is happening on purpose. But then I don't know what to do about this.
Should this unique string be stored in the PropertyXLink?

So for instance how do I set the value of the PropertyXLink? Is it like : 
```
    def addSelection(self, doc_name, obj_name, sub_name, mousePos):
        obj = doc.getObject(obj_name)
        joint.Reference = [obj, [sub_name] # Reference is a PropertyXLinkSubHidden
```

bgbsww, 2024-07-16:
That looks correct to me; note that the tail end of the sub_name is a dot followed by the short name of the element.  I would try it the way you proposed, and if that did not work correctly look into just using the part after the last dot.

In c++ that is Data::oldElementName(subname) in ElementNamingUtils.cpp.  Looks like getOldElementName() in a DocumentObserver might get at this.    At python level, looks like resolveSubElement in a DocumentObject might do this.

Something like obj.resolveSubElement(sub_name)[2] to get the short name.

bgbsww, 2024-07-16:
> But apparantly only for PartDesign bodies. For others objects (box for example) then it's ok.

A simple box does not have an elementmap, and would not have the longer names.  Take a box and a cylinder ( or any other primitive) and run a boolean on them ( Cut, Fuse, whatever ), and the resulting object will have and elementMap, and will probably show the same long names in selection - it shouldn't be PartDesign specific, it's just that the PartDesign objects are generally more complicated things built out of other things - like by padding a sketch, for example.

PaddleStroke, 2024-07-17:
Thanks for the help, resolveSubElement is very helpful.
So this is not a bug afterall but intended. So I'll close this issue.

closed 2024-07-17 by PaddleStroke

PaddleStroke, 2024-07-19:
It appears that in my tests my selection observer was using the NoResolve flag... So it appears the problem was between the chair and the keyboard.