FreeCAD/FreeCAD #15436: Import IFC: <class 'TypeError'>: 'NoneType' object is not iterable

Mannshoch, 2024-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Open a certain IFC not work. 
I'm not allowed to upload the IFC.
The IFC was exported by Solidworks 2023

```
15:23:25  pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "/snap/freecad/1077/usr/Mod/BIM/nativeifc/ifc_import.py", line 52, in open
    insert(filename, doc.Name, singledoc=True)
  File "/snap/freecad/1077/usr/Mod/BIM/nativeifc/ifc_import.py", line 97, in insert
    ifc_psets.load_psets(prj_obj)
  File "/snap/freecad/1077/usr/Mod/BIM/nativeifc/ifc_psets.py", line 232, in load_psets
    load_psets(child)
  File "/snap/freecad/1077/usr/Mod/BIM/nativeifc/ifc_psets.py", line 232, in load_psets
    load_psets(child)
  File "/snap/freecad/1077/usr/Mod/BIM/nativeifc/ifc_psets.py", line 232, in load_psets
    load_psets(child)
  [Previous line repeated 1 more time]
  File "/snap/freecad/1077/usr/Mod/BIM/nativeifc/ifc_psets.py", line 226, in load_psets
    show_psets(obj)
  File "/snap/freecad/1077/usr/Mod/BIM/nativeifc/ifc_psets.py", line 90, in show_psets
    psets = get_psets(element)
  File "/snap/freecad/1077/usr/Mod/BIM/nativeifc/ifc_psets.py", line 59, in get_psets
    for prop in pset.HasProperties:
<class 'TypeError'>: 'NoneType' object is not iterable
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38166 (Git) Snap 1077
Build type: Release
Branch: main
Hash: 4bb6d191b4686919a2f247bcfce6c186703a3d86
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Mannshoch, 2024-07-15:
I'm not sure what you mean with 'Striped down version'

I assume you mean the IFC file version
The header of the file:
```
ISO-10303-21;
HEADER;
FILE_DESCRIPTION (( 'IFC4' ),'2;1' );
FILE_NAME ('Haus.IFC',
     '2023-06-06T11:06:02',
    ( '' ),
    ( '' ),
    'SwIFC',
    'SolidWorks 2023',
    '' );
FILE_SCHEMA (( 'IFC4' ));
ENDSEC;

DATA;
```

yorikvanhavre, 2024-07-15:
It's not possible to know what the exact problem is without seeing the file, but here is a fix anyway

closed 2024-07-16 by yorikvanhavre

Mannshoch, 2024-07-16:
Schould I test something?
If you give me advise I try to be a help.

Mannshoch, 2024-07-18:
In this version, I have still the same error:
```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38259 (Git) Snap 1080
Build type: Release
Branch: main
Hash: c8046b1c60c786caca3050cf51f853454ffb5388
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
```

yorikvanhavre, 2024-07-19:
I am not sure that version has the fix already...

Mannshoch, 2024-07-19:
Today I updated to 
```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38260 (Git) Snap 1081
Build type: Release
Branch: main
Hash: [ff61f56d19f65dcb9b8cb4b4768d54c28346c359](https://github.com/FreeCAD/FreeCAD/commit/ff61f56d19f65dcb9b8cb4b4768d54c28346c359)
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
```
I still have the same Problem. 

I assume there is something wrong with the patch.

yorikvanhavre, 2024-07-30:
Can you re-run the test and past the error message? The details might have changed and another similar error appeared elsewhere

Mannshoch, 2024-07-30:
Thanks for your engagement. I'm currently on vacation. I will test it as soon as possible.

Mannshoch, 2024-08-05:
```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38387 (Git) Snap 1098
Build type: Release
Branch: main
Hash: c0c7bd647dc91f8807d8632585b6e6a53a040798
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
```
[image]

Mannshoch, 2024-08-12:
I refreshed my installation. Now the errors changed.

```
09:58:32  pyException: Traceback (most recent call last):
  File "/snap/freecad/1102/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "/snap/freecad/1102/usr/Mod/BIM/nativeifc/ifc_generator.py", line 68, in generate_geometry
    shape, colors = generate_shape(ifcfile, elements, cached)
  File "/snap/freecad/1102/usr/Mod/BIM/nativeifc/ifc_generator.py", line 126, in generate_shape
    iterator = get_geom_iterator(ifcfile, elements, brep_mode=True)
  File "/snap/freecad/1102/usr/Mod/BIM/nativeifc/ifc_generator.py", line 365, in get_geom_iterator
    settings.set(settings.DISABLE_TRIANGULATION, True)
  File "/snap/freecad/1102/lib/python3.10/site-packages/ifcopenshell/geom/main.py", line 212, in __getattr__
    raise AttributeError("'Settings' object has no attribute '%s'" % k)
<class 'AttributeError'>: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
09:58:32  Haus: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
09:58:32  Imported Haus.IFC ( 4.01 Mb ) in 00:00
```

Mannshoch, 2024-08-13:
This error I also get with this IFC:
http://www.ifcwiki.org/images/0/09/KIT-Bridge.zip

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38460 (Git) Snap 1103
Build type: Release
Branch: main
Hash: aef3bdfe679d7965ee13192e1fc160b18f7e6826
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
```

```
10:38:15  Note: API not available due to missing dependencies: geometry.add_door_representation - No module named 'mathutils'
10:38:15  Note: API not available due to missing dependencies: geometry.add_railing_representation - No module named 'mathutils'
10:38:15  Note: API not available due to missing dependencies: geometry.add_representation - No module named 'bpy'
10:38:15  Note: API not available due to missing dependencies: geometry.add_window_representation - No module named 'mathutils'
10:38:15  Note: API not available due to missing dependencies: grid.create_axis_curve - No module named 'mathutils'
10:38:15  pyException: Traceback (most recent call last):
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 68, in generate_geometry
    shape, colors = generate_shape(ifcfile, elements, cached)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 126, in generate_shape
    iterator = get_geom_iterator(ifcfile, elements, brep_mode=True)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 365, in get_geom_iterator
    settings.set(settings.DISABLE_TRIANGULATION, True)
  File "/snap/freecad/1103/lib/python3.10/site-packages/ifcopenshell/geom/main.py", line 212, in __getattr__
    raise AttributeError("'Settings' object has no attribute '%s'" % k)
<class 'AttributeError'>: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  pyException: Traceback (most recent call last):
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 68, in generate_geometry
    shape, colors = generate_shape(ifcfile, elements, cached)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 126, in generate_shape
    iterator = get_geom_iterator(ifcfile, elements, brep_mode=True)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 365, in get_geom_iterator
    settings.set(settings.DISABLE_TRIANGULATION, True)
  File "/snap/freecad/1103/lib/python3.10/site-packages/ifcopenshell/geom/main.py", line 212, in __getattr__
    raise AttributeError("'Settings' object has no attribute '%s'" % k)
<class 'AttributeError'>: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  pyException: Traceback (most recent call last):
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 68, in generate_geometry
    shape, colors = generate_shape(ifcfile, elements, cached)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 126, in generate_shape
    iterator = get_geom_iterator(ifcfile, elements, brep_mode=True)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 365, in get_geom_iterator
    settings.set(settings.DISABLE_TRIANGULATION, True)
  File "/snap/freecad/1103/lib/python3.10/site-packages/ifcopenshell/geom/main.py", line 212, in __getattr__
    raise AttributeError("'Settings' object has no attribute '%s'" % k)
<class 'AttributeError'>: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  pyException: Traceback (most recent call last):
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 68, in generate_geometry
    shape, colors = generate_shape(ifcfile, elements, cached)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 126, in generate_shape
    iterator = get_geom_iterator(ifcfile, elements, brep_mode=True)
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_generator.py", line 365, in get_geom_iterator
    settings.set(settings.DISABLE_TRIANGULATION, True)
  File "/snap/freecad/1103/lib/python3.10/site-packages/ifcopenshell/geom/main.py", line 212, in __getattr__
    raise AttributeError("'Settings' object has no attribute '%s'" % k)
<class 'AttributeError'>: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  pyException: Traceback (most recent call last):
  File "/snap/freecad/1103/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
[... 434 lines cut ...]
10:38:15  _IfcSlab001: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Pile-2-3: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Pile-2-1: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Pile-2-5: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Pile-2-6: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Pile-1-5: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
[... cut ...]
10:38:15  Column-02: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Vibration Damper 02 (viscous): 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Vibration Damper 03 (viscous): 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Vibration Isolator 02: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Vibration Isolator 03: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
10:38:15  Imported KIT-Bridge.ifc ( 1.75 Mb ) in 00:01
```

Mannshoch, 2024-08-22:
@yorikvanhavre Should I create a new Issue for the `DISABLE_TRIANGULATION` problem?

PR #15444 (2024-07-16): BIM: Fixed IFC import with None psets
Fixes #15436
PR #15773 (2024-08-05): BIM: Fixed nativeIFC psets loading - 
Better fix for #15436
Fixes #15436
commit 33088e00 (2024-08-05): BIM: Fixed nativeIFC psets loading - fixes #15436