FreeCAD/FreeCAD #15438: Part Design features turning Transparency Off

leoheck, 2024-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Part Design features are changing the opacity of the model.
Opacity should remain the same or even better, activated to be a bit transparent.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git) AppImage
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.22
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-15:
Is this also the case in the stable 0.21.2 version?
Where did you apply transparency, on a feature, then added another feature? On the body? On a parent Part container,...?
Would be great if you could ad steps to reproduce and an example file. TIA

leoheck, 2024-07-15:
It is also overriding the Draw style.
[image]

> Is this also the case in the stable 0.21.2 version?

I am not sure, I don't use 0.21.2
I think it started one or 2 weekly versions ago.

> Where did you apply transparency, on a feature, then added another feature? On the body? On a parent Part container,...?

I apply it to the Body itself. When I reapply the transparency and then edit an existing feature this does not happen. When I create a new one, it does.

> Would be great if you could ad steps to reproduce and an example file. TIA

I made a new project here to reproduce this issue, it still happens, but not it is a bit different again.
Steps:

1. Create a Body with Part Design Workbench
2. Create a sketch (I am creating a rectangle)
3. Pad it.
4. Toggle Body transparency with V, T
5. Select a face of the sketch (I am using the top face)
6. Make another shape (I am using another rectangle)
7. Pad it (on my report the transparency was being disabled here, but not on a new project file)
8. Apply the pad (this example disables transparency here)

[transparency_issue.zip](https://github.com/user-attachments/files/16235834/transparency_issue.zip)

leoheck, 2024-07-15:
> It is also overriding the Draw style.

When this happens, setting the mode `As is` again does not update the models.. I have to toggle it, using another mode, and then setting the `As is` in the end.

maxwxyz, 2024-07-15:
@davesrocketshop was there a recent change in Materials / transparency handling in PartDesign?

davesrocketshop, 2024-07-15:
Not by me. Your last merges I believe were it.

davesrocketshop, 2024-07-15:
Sorry.. by @wwmayer

leoheck, 2024-08-01:
This still happening on `0.22.0dev.38334 (Git) AppImage`

Steps to reproduce:
1. Create a Part Design Body
2. Sketch with a rectangle in the middle of it.
3. Pad it (let's say 100 mm)
4. Click the Body, and type `v` `t` to activate the transparency (so you can see hidden edges through it)
5. Select a face, let's say the one in front of you (X-Z axis)
6. Create a Sketch of the selected face
7. Create something on the object, let's say another rectangle.
8. Pad it. 

Step 8 will launch the Pad properties and remove the transparency that should be preserved. (annoying)
If I hit Ok to make the Pad, it will keep the transparency off. (annoying)
If I hit Cancel, it does not pad but also it keeps the existing transparency.

I am sharing this example here so it is easier to play with the issue.
[transparency_playground.zip](https://github.com/user-attachments/files/16457637/transparency_playground.zip)

leoheck, 2024-08-01:
The transparency does not change when editing and existing Feuature. It looks like this issue happens when the Feature is being created.

amon-sha, 2024-08-11:
Adding a feature completely resets the shape appearance, not only the transparency.
The regression has been introduced in this commit: 8b5a3b1124c06228266b5cc06dceb64db61718cd
There is a call to `copyColorsfromTip` method in `ViewProviderBody::updateData` (`src/Mod/PartDesign/Gui/ViewProviderBody.cpp`) which looks unnecessary to me.
Removing the method call results in intended behavior: body's shape appearance is preserved when a feature is added.
I'm not sure if this is a correct fix though.

maxwxyz, 2024-08-11:
@davesrocketshop is this actually a duplicate of https://github.com/FreeCAD/FreeCAD/issues/15503 ?

davesrocketshop, 2024-08-11:
> @davesrocketshop is this actually a duplicate of #15503 ?

Yes

maxwxyz, 2024-08-12:
Duplicate of #15503

closed 2024-08-12 by maxwxyz