FreeCAD/FreeCAD #15442: Sketcher does not honor V,# shortcuts

leoheck, 2024-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be nice to be able to `Toggle Mode` with shortcuts when editing a Sketch, especially since the transparency mode is not working properly. 

However if in Sketch Edit mode, I cannot use `v,3` and `v,1`. Sketcher ignores (or maybe consumes) the `v` and applies the `1` and `3` which is the shortcut to change the Standard view.

Not sure if this matters but my Navigation Style is Touchpad.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git) AppImage
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.22
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-15:
In Sketcher Edit Mode, the keyboard shortcut `V` has been `Constraint Vertical` since at least the 0.18.x versions and probably before that so IMHO that isn't going to change when so many users have got accustomed to using it.

leoheck, 2024-07-15:
Yeah, the Draw Mode can be improved then.