FreeCAD/FreeCAD #15452: Pocket through revolve creates a "skin" of the pocket on one side

gitnoel, 2024-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When pocketing a circle through a revolve it produces an 'extruded skin' on one side of the pocket.
I tried with a second body, creating pad of a circle, and pocketing another sketch through it and it worked okay.

### Full version info

```shell
OS: Linux Mint 21 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git) AppImage
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/New Zealand (en_NZ)
Installed mods: 
  * Help 1.0.3
  * fcgear 1.0.0
  * ExplodedAssembly
  * Curves 0.6.41
  * lattice2 1.0.0
  * FeedsAndSpeeds 0.5.0
  * sheetmetal 0.2.59
  * InventorLoader 1.4.0
```

### Subproject(s) affected?

None

### Anything else?

**ReportView**
10:18:57  Cannot find icon: Sketcher_Pointer_Create_Ellipse
10:19:17  Reading failed from embedded file: StringHasher.Table.txt (1403 bytes, 293 bytes compressed)
10:19:17  <ElementMap> <string>(1): No hasherRef
10:19:17  <ElementMap> <string>(1): No hasherRef
10:19:17  <ElementMap> <string>(1): No hasherRef
10:19:17  <ElementMap> <string>(1): No hasherRef
10:19:17  <ElementMap> <string>(1): No hasherRef

**PythonConsole - I replaced hash with H so the formatting stays sensible**
Python 3.11.9 | packaged by conda-forge | (main, Apr 19 2024, 18:53:30) [GCC 12.3.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> H Gui.runCommand('Std_Workbench',1)
>>> H Gui.runCommand('Std_ViewStatusBar',1)
>>> HHH Begin command Std_Open
>>> FreeCAD.openDocument('/home/mee-cee/Documents/3D Printing/FreeCad/SinkPlug.FCStd')
>>> H App.setActiveDocument("SinkPlug")
>>> H App.ActiveDocument=App.getDocument("SinkPlug")
>>> H Gui.ActiveDocument=Gui.getDocument("SinkPlug")
>>> HHH End command Std_Open
>>> H Gui.runCommand('Std_About',0)
>>> 
[plug.zip](https://github.com/user-attachments/files/16241049/plug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-16:
This could be an OCC bug and is also present in the stable 0.21.2 version. Could you create a topic on the forums, I'm sure this is a known issue.

gitnoel, 2024-07-16:
Sure I'll create a topic.  This is the first time I've struck this
particular issue, maybe I've been fortunate?.  I don't know what an OCC bug
is, sorry.

luzpaz, 2024-08-25:
@gitnoel do you mind attaching a screenshot to this ticket to illustrate the issue ?

gitnoel, 2024-09-03:
@luzpaz Image attached, apologies for the delay
[image]

gitnoel, 2024-12-09:
As of this weekly build (below), the issue is still present

OS: Linux Mint 21 (MATE/mate/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/New Zealand (en_NZ)
Stylesheet/Theme/QtStyle: unset/Light theme/Qt default
Installed mods: 
  * Help 1.0.3
  * fcgear 1.0.0
  * ExplodedAssembly
  * Curves 0.6.41
  * lattice2 1.0.0
  * FeedsAndSpeeds 0.5.0
  * sheetmetal 0.2.59
  * InventorLoader 1.4.0

FlachyJoe, 2024-12-09:
Can not reproduce here

> OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
> Architecture: x86_64
> Version: 1.1.0dev.39516 +12 (Git)
> Build type: Release
> Branch: dev
> Hash: 562e4c00ea4e0d8b4bee3b92261f99acf9bd31fe
> Python 3.12.7, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, **OCC 7.8.2.dev**
> Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
> Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

[image]

Maybe an OCC version specific issue.

The test file: 
[RevAndPocket.FCStd.zip](https://github.com/user-attachments/files/18058291/RevAndPocket.FCStd.zip)

gitnoel, 2025-10-05:
Issue is still present in 1.02.  
A workaround has been to offset the circle center by 0.001mm vertically

OS: Linux Mint 22.2 (MATE/mate/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/New Zealand (en_NZ)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/Fusion
Installed mods: 
  * sheetmetal 0.7.22
  * fasteners 0.5.38

stefano2734, 2026-09-12:
Occt 7.8.x is quite old here.
Should be tested in 26.3 dev with occt 7.9.3 and 8.0.1 to see the actual behavior.

gitnoel, 2026-09-13:
Still present as of 26.3.0Dev FreeCAD_weekly-2026.09.09-Linux-x86_64.AppImage
My workaround is still offset the circle center by 0.001mm vertically

OS: Linux Mint 22.2 (MATE/mate/xcb)
Architecture: x86_64
Version: 26.3.0dev.20260908 (Git shallow) AppImage
Build date: 2026/09/08 23:17:32
Build type: Release
Branch: (HEAD detached at 306259d7a)
Hash: 306259d7aa3e7f3b40a22232eb60973deb39ca20
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/New Zealand (en_NZ)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 25.2.8-0ubuntu0.24.04.2/Intel/Mesa Intel(R) HD Graphics 4600 (HSW GT2)
Installed mods:
  * Fasteners Workbench 0.5.64
  * Feeds & Speeds 0.6.0
  * SheetMetal Workbench 0.8.22
  * dxf-library

stefano2734, 2026-09-13:
> Still present as of 26.3.0Dev FreeCAD_weekly-2026.09.09-Linux-x86_64.AppImage My workaround is still offset the circle center by 0.001mm vertically
> 
> OS: Linux Mint 22.2 (MATE/mate/xcb) Architecture: x86_64 Version: 26.3.0dev.20260908 (Git shallow) AppImage Build date: 2026/09/08 23:17:32 Build type: Release Branch: (HEAD detached at [306259d](https://github.com/FreeCAD/FreeCAD/commit/306259d7aa3e7f3b40a22232eb60973deb39ca20)) Hash: [306259d](https://github.com/FreeCAD/FreeCAD/commit/306259d7aa3e7f3b40a22232eb60973deb39ca20) Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3 shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.0, 

OCC 7.8.1
> 
> * Fasteners Workbench 0.5.64
> * Feeds & Speeds 0.6.0
> * SheetMetal Workbench 0.8.22
> * dxf-library

What should change with old occt 7.8.1 in your installation By an occt problem?