FreeCAD/FreeCAD #15453: Crash when exiting task with nothing entered in number field

flaviut, 2024-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a sketch with a rectangle on it (or anything else) in Part Design
2. Start a pad. Delete the number in the pad task dialog
3. Press ESC to close the pad task. Observe the segfault.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3cae0) [0x7d4f2c250ae0]
#1  0x7d4edc54cedb in PartDesignGui::TaskExtrudeParameters::onLengthChanged(double) from /tmp/.mount_freecabUtBMZ/usr/lib/PartDesignGui.so+0xb
#2  /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7d4f2c6eb6f2]
#3  0x7d4f2f827294 in Gui::QuantitySpinBox::valueChanged(double) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x44
#4  0x7d4f2f8281ea in Gui::QuantitySpinBox::updateFromCache(bool, bool) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0xda
#5  0x7d4f2f82fa24 in Gui::QuantitySpinBox::focusOutEvent(QFocusEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x94
#6  0x7d4f2d5af53b in QWidget::event(QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0xaab
#7  0x7d4f2d5703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#8  0x7d4f2f33cf69 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x99
#9  0x7d4f2c6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Core.so.5+0x108
#10  0x7d4f2d5742e0 in QApplicationPrivate::setFocusWidget(QWidget*, Qt::FocusReason) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0x170
#11  0x7d4f2d5a9867 in QWidget::setFocus(Qt::FocusReason) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0x107
#12  0x7d4f2f6c674e in Gui::TaskView::TaskView::updateWatcher() from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x1ce
#13  0x7d4f2f6c5b35 in Gui::TaskView::TaskView::addTaskWatcher() from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x95
#14  0x7d4f2f6c5db6 in Gui::TaskView::TaskView::removeDialog() from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x166
#15  /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7d4f2c6eb6f2]
#16  /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5(+0x2fab70) [0x7d4f2d6fab70]
#17  /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7d4f2c6eb964]
#18  0x7d4f2d65b522 in QAbstractButton::clicked(bool) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0x32
#19  /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5(+0x25b73a) [0x7d4f2d65b73a]
#20  0x7d4f2d65d5a9 in QAbstractButton::click() from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0x89
#21  0x7d4f2f6c5756 in Gui::TaskView::TaskView::keyPressEvent(QKeyEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x326
#22  0x7d4f2d5afa00 in QWidget::event(QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0xf70
#23  0x7d4f2d656c6e in QFrame::event(QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#24  0x7d4f2d5703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#25  0x7d4f2d577bdf in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0xd1f
#26  0x7d4f2f33cf69 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x99
#27  0x7d4f2c6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Core.so.5+0x108
#28  /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca118) [0x7d4f2d5ca118]
#29  0x7d4f2d5703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#30  0x7d4f2f33cf69 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x99
#31  0x7d4f2c6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Core.so.5+0x108
#32  0x7d4f2cd312fb in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Gui.so.5+0xdb
#33  0x7d4f2cd0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Gui.so.5+0xac
#34  /tmp/.mount_freecabUtBMZ/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7d4f271d2bea]
#35  /tmp/.mount_freecabUtBMZ/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7d4f2af10ad7]
#36  /tmp/.mount_freecabUtBMZ/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7d4f2af13ac7]
#37  /tmp/.mount_freecabUtBMZ/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7d4f2af140ce]
#38  0x7d4f2c70aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Core.so.5+0x66
#39  0x7d4f2c6b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Core.so.5+0x11b
#40  0x7d4f2c6bd212 in QCoreApplication::exec() from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libQt5Core.so.5+0x82
#41  0x7d4f2f29f6e2 in Gui::Application::runApplication() from /tmp/.mount_freecabUtBMZ/usr/bin/../lib/libFreeCADGui.so+0x972
#42  /tmp/.mount_freecabUtBMZ/usr/bin/freecad(+0x7620) [0x5be110bdc620]
#43  /usr/lib/libc.so.6(+0x25c88) [0x7d4f2c239c88]
#44  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7d4f2c239d4c]
#45  /tmp/.mount_freecabUtBMZ/usr/bin/freecad(+0x7981) [0x5be110bdc981]
```

### Full version info

```shell
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git) AppImage
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
[... 69 lines cut ...]
#8  0x75e0ab73cf69 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecafiGzmu/usr/bin/../lib/libFreeCADGui.so+0x99
#9  0x75e0a8ab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecafiGzmu/usr/bin/../lib/libQt5Core.so.5+0x108
#10  0x75e0a99742e0 in QApplicationPrivate::setFocusWidget(QWidget*, Qt::FocusReason) from /tmp/.mount_freecafiGzmu/usr/bin/../lib/libQt5Widgets.so.5+0x170
```

## Fillet

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3cae0) [0x7f3943250ae0]
#1  0x7f38e85684aa in PartDesignGui::TaskDressUpParameters::getBase() const from /tmp/.mount_freecag1fIga/usr/lib/PartDesignGui.so+0x1a
#2  0x7f38e856851f in PartDesignGui::TaskDressUpParameters::showObject() from /tmp/.mount_freecag1fIga/usr/lib/PartDesignGui.so+0xf
#3  0x7f38e85685c9 in PartDesignGui::TaskDressUpParameters::setSelectionMode(PartDesignGui::TaskDressUpParameters::selectionModes) from /tmp/.mount_freecag1fIga/usr/lib/PartDesignGui.so+0x29
#4  0x7f38e855c3a7 in PartDesignGui::TaskFilletParameters::onLengthChanged(double) from /tmp/.mount_freecag1fIga/usr/lib/PartDesignGui.so+0x17
#5  /tmp/.mount_freecag1fIga/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f39436eb6f2]
#6  0x7f3946827294 in Gui::QuantitySpinBox::valueChanged(double) from /tmp/.mount_freecag1fIga/usr/bin/../lib/libFreeCADGui.so+0x44
#7  0x7f39468281ea in Gui::QuantitySpinBox::updateFromCache(bool, bool) from /tmp/.mount_freecag1fIga/usr/bin/../lib/libFreeCADGui.so+0xda
#8  0x7f394682fa24 in Gui::QuantitySpinBox::focusOutEvent(QFocusEvent*) from /tmp/.mount_freecag1fIga/usr/bin/../lib/libFreeCADGui.so+0x94
#9  0x7f39445af53b in QWidget::event(QEvent*) from /tmp/.mount_freecag1fIga/usr/bin/../lib/libQt5Widgets.so.5+0xaab
#10  0x7f39445703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecag1fIga/usr/bin/../lib/libQt5Widgets.so.5+0x7e
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-07-16:
Confirmed on:

**OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * Curves 0.6.40
**

maxwxyz, 2024-07-16:
No crash but error on Windows11. 
No issue in the stable 0.21.2 version.

wwmayer, 2024-07-19:
The problems are caused by these changes: bf9584a90b and 8baf75af845

An easy fix to avoid the crash is to change `TaskView::updateWatcher` by removing the lines
```
    if (ActiveCtrl || ActiveDialog)
        return;
```

The reason of the crash is that the edit mode of the document is reset before the spin box loses the focus. The PD task panel receives a signal from the spin box but at this point the edited object is already unexpectedly destroyed which then leads to the crash.

With the above change the old order of execution is restored: the spin box emits a signal before the edit mode is reset so that when the PD task panel gets the signal the edited object is still alive.

Nonetheless, it's recommended to also merge #15499 because this makes the code more robust to correctly handle the eventual destruction of the edited object.

PR #15499 (2024-07-22): Fix possible crashes in PD task panels
* Fix possible crashes in dress-up task panels
* Fix possible crash in primitive dialog
* Add template helper functions to safely access feature or view provider
* In subclasses of TaskFeatureParameters use the new helper functions and check for null pointer.
* Replace helper functions in subclasses of TaskDlgFeatureParameters with a template function

This fixes #15453
commit 5731a276 (2024-07-22): PD: Add template helper functions to safely access feature or view provider

In subclasses of TaskFeatureParameters use the new helper functions and check for null pointer. This fixes #15453
closed 2024-07-22 by chennes