FreeCAD/FreeCAD #15462: Whitescreen workarea when opening any freecad file

vini-nu, 2024-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Whitescreen workarea when opening any cad files. 

### Full version info

```shell
[code]
OS: EndeavourOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
[/code]

Wayland Arch Linux installed through Arch repos
```

### Subproject(s) affected?

None

### Anything else?

Freecad STDOUT on launch:
```
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
#####                 ####  ###   ####
#                    #      # #   #   #
#     ##  #### ####  #     #   #  #   #
####  # # #  # #  #  #     #####  #   #
#     #   #### ####  #    #     # #   #
#     #   #    #     #    #     # #   #  ##  ##  ##
#     #   #### ####   ### #     # ####   ##  ##  ##

Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/lib/freecad/bin/qtwebengine_dictionaries'
Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/lib/qt6/qtwebengine_dictionaries'
Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/lib/qt6/qtwebengine_dictionaries'
```

And when trying to open file freecad STDOUT outputs:
```
Coin warning in cc_glglue_instance(): Error when setting up the GL context. This can happen if there is no current context, or if the context has been set up incorrectly.
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
Cannot create object 'Page': (No module named 'Drawing')
Cannot create object 'View': (No module named 'Drawing')
Cannot create object 'View001': (No module named 'Drawing')
Cannot create object 'View002': (No module named 'Drawing')
Cannot create object 'Annotation': (No module named 'Drawing')
Cannot create object 'Annotation001': (No module named 'Drawing')
Cannot create object 'Annotation002': (No module named 'Drawing')
Cannot create object 'ViewGroup002': (No module named 'Drawing')
Cannot create object 'ViewGroup004': (No module named 'Drawing')
/usr/lib/freecad/Mod/Draft/draftguitools/gui_trackers.py:426: SyntaxWarning: invalid escape sequence '\['
pts = re.findall("point \[(.*?)\]", buf)[0]
/usr/lib/freecad/Mod/Draft/draftguitools/gui_trackers.py:504: SyntaxWarning: invalid escape sequence '\['
pts = re.findall("point \[(.*?)\]", buf)[0]
/usr/lib/freecad/Mod/Draft/draftguitools/gui_trackers.py:639: SyntaxWarning: invalid escape sequence '\['
pts = re.findall("point \[(.*?)\]", buf)[0]
/usr/lib/freecad/Mod/Arch/ArchSectionPlane.py:751: SyntaxWarning: invalid escape sequence '\#'
marker = re.findall("<line x1=.*?stroke=\"\#ffffff\".*?\/>",svg)
/usr/lib/freecad/Mod/Arch/ArchSectionPlane.py:767: SyntaxWarning: invalid escape sequence '\#'
svg = re.sub("<line x1=.*?stroke=\"\#ffffff\".*?\/>","",svg,count=1)
/usr/lib/freecad/Mod/Arch/ArchSectionPlane.py:795: SyntaxWarning: invalid escape sequence '\?'
svg = re.sub("<\?xml.*?>","",svg,flags=re.MULTILINE|re.DOTALL)
/usr/lib/freecad/Mod/Arch/ArchSectionPlane.py:797: SyntaxWarning: invalid escape sequence '\/'
svg = re.sub("<\/svg>","",svg,flags=re.MULTILINE|re.DOTALL)
/usr/lib/freecad/Mod/Draft/importSVG.py:416: SyntaxWarning: invalid escape sequence '\.'
_num = '([-+]?[0-9]*\.?[0-9]+([eE][-+]?[0-9]+)?)'
/usr/lib/freecad/Mod/Draft/importSVG.py:732: SyntaxWarning: invalid escape sequence '\d'
inks_ver_pars = re.search("\d+\.\d+", inks_full_ver)
/usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
_command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
/usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
_command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
/usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
_command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
/usr/lib/freecad/Mod/Draft/importSVG.py:938: SyntaxWarning: invalid escape sequence '\.'
_num = '[-+]?[0-9]*\.?[0-9]+'
/usr/lib/freecad/Mod/Draft/importSVG.py:1620: SyntaxWarning: invalid escape sequence '\('
_val = '\((.*?)\)'
/usr/lib/freecad/Mod/Draft/importSVG.py:1621: SyntaxWarning: invalid escape sequence '\s'
_transf = _op + '\s*?' + _val
Traceback (most recent call last):
File "/usr/lib/freecad/Mod/Arch/ArchWall.py", line 1620, in attach
image = Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
^^^^^^^^^^^^^^^^^^^
File "/usr/lib/freecad/Mod/Draft/draftutils/utils.py", line 789, in svg_patterns
loadSvgPatterns()
File "/usr/lib/freecad/Mod/Draft/draftutils/utils.py", line 763, in load_svg_patterns
p = importSVG.getContents(file, 'pattern')
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/freecad/Mod/Draft/importSVG.py", line 1709, in getContents
f = pythonopen(filename)
^^^^^^^^^^
<class 'NameError'>: name 'pythonopen' is not defined
Traceback (most recent call last):
File "/usr/lib/freecad/Mod/Arch/ArchWall.py", line 1714, in setDisplayMode
self.fset.coordIndex.deleteValues(0)
^^^^^^^^^
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
Traceback (most recent call last):
File "/usr/lib/freecad/Mod/Arch/ArchWall.py", line 1714, in setDisplayMode
self.fset.coordIndex.deleteValues(0)
^^^^^^^^^
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
Coin warning in SoGLPolygonOffsetElement::updategl(): OpenGL driver doesn't support z-buffer offsetting
SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
Traceback (most recent call last):
File "/usr/lib/freecad/Mod/Draft/draftobjects/array.py", line 70, in onDocumentRestored
self.update_properties_0v21(obj)
File "/usr/lib/freecad/Mod/Draft/draftobjects/array.py", line 74, in update_properties_0v21
self.execute(obj) # Required to update Count to the correct value.
^^^^^^^^^^^^^^^^^
File "/usr/lib/freecad/Mod/Draft/draftobjects/array.py", line 423, in execute
obj.IntervalZ,
^^^^^^^^^^^^^
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'IntervalZ'
```

Result:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-07-16:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/13893

maxwxyz, 2024-07-16:
Duplicate of #13893

closed 2024-07-16 by maxwxyz