FreeCAD/FreeCAD #15463: Freecad file opened twice.

leoheck, 2024-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When recovering from a crash, freecad tries to fix files that were open during the crash.

However, if I launch Freecad from the command line (which I always do) by passing the filename to be opened, it can open the same file twice.

```
<Gui> Document.cpp(1066): Physical path: /tmp/Internal-Chamber.FCStd
<Gui> Document.cpp(1068):   Document: Internal-Chamber (Internal_Chamber): /tmp/Internal-Chamber.FCStd
<Gui> Document.cpp(1068):   Document: Internal-Chamber (Internal_Chamber1): /tmp/Internal-Chamber.FCStd
<Gui> Document.cpp(1066): Physical path: /tmp/Internal-Chamber.FCStd
<Gui> Document.cpp(1068):   Document: Internal-Chamber (Internal_Chamber): /tmp/Internal-Chamber.FCStd
<Gui> Document.cpp(1068):   Document: Internal-Chamber (Internal_Chamber1): /tmp/Internal-Chamber.FCStd
<Gui> Document.cpp(1066): Physical path: /tmp/Internal-Chamber.FCStd
<Gui> Document.cpp(1068):   Document: Internal-Chamber (Internal_Chamber): /tmp/Internal-Chamber.FCStd
<Gui> Document.cpp(1068):   Document: Internal-Chamber (Internal_Chamber1): /tmp/Internal-Chamber.FCStd
```

It could prevent that by checking if the file to be opened is in the list of files to be fixed.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git) AppImage
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-16:
Is this also happening with the stable version or is this bug also present there?

leoheck, 2024-07-17:
Probably. Do you know how to trigger file the file recovery? I can test on 0.21 if can can trigger that state somehow.

krushia, 2024-07-18:
Easiest way is to open a file, edit it without saving, wait for FreeCAD to do an autosave, then `killall -9 freecad`.

The actual recovery mechanism is directories in `~/.cache/FreeCAD/Cache` (assuming default/unset `$XDG_CACHE_HOME`).

maxwxyz, 2024-07-28:
any news on this @leoheck

leoheck, 2024-07-28:
What do you mean? When you launch Freecad from the command line passing a file (and maybe from the GUI when you double click on a file), if the file has to be repaired since Freecad broke the last launch with the same file being edited, it is going to launch that file twice. And this cloud be prevented.

maxwxyz, 2024-07-28:
Is this also opening twice with the stable version?

leoheck, 2024-12-05:
I am closing this issue since I did not see the problem anymore, and when I had a similar problem Freecad did not open the file since the other file was newer than the cached file.

closed 2024-12-05 by leoheck