FreeCAD/FreeCAD #15464: Cannot place ShapeString

leoheck, 2024-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I could not place the `Draft > ShapeText`
Also, the coordinates are crazy.

The bottom-left panel shows the current point, and the top-right shows the coordinates gotten by the Draft ShapeString. The Y is in `Km`.
[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git) AppImage
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-07-16:
Y is in Km when the facing XZ
When facing YZ, the km goes on the X field 
When facing XY it is ok.

But still does not place the StringText

leoheck, 2024-07-16:
Ah, It did not update the screen. I have to hit Ok, for it to create the object.

leoheck, 2024-07-16:
Also, the Font field always comes empty when using the ShapeString.

Syres916, 2024-07-16:
> Also, the Font field always comes empty when using the ShapeString.
> 

Yes, if you've not filled in the _Preferences_ setting which can be a folder or a font file (I personally have a folder set) https://wiki.freecad.org/Draft_ShapeString#Preferences

maxwxyz, 2024-07-16:
So is there still a remaining issue?

Roy-043, 2024-07-17:
This is really a Help request. User has to set the working plane appropriately. Closing this.

closed 2024-07-17 by Roy-043

leoheck, 2024-07-17:
> > Also, the Font field always comes empty when using the ShapeString.
> 
> Yes, if you've not filled in the _Preferences_ setting which can be a folder or a font file (I personally have a folder set) https://wiki.freecad.org/Draft_ShapeString#Preferences

I set a font when I used the first time.
The field could remember that.

But it could also come with a standard font. 
If there is a standard font. When user click the field to change it, he will know better where to find others.

leoheck, 2024-07-17:
> This is really a Help request. User has to set the working plane appropriately. Closing this.

But how. When I click the face I want to place a font one of the coordinates are in Km, doesn't make any sense.

Also, the text is always placed on xy

leoheck, 2024-07-17:
I am going to summarize the issues.

- No matter the face I am clicking on my object the X or Y is in Km not mm.
- The object is placed on XY not on ZY or ZX
- The text is visible only after hitting Ok. So no way of tunning the placement while creating the object 
- The font is an blank field (always). You can set the loved font on settings page, but it should come with a font set.

Roy-043, 2024-07-19:
1+2: RTM (if you need more help start a topic in the forum).
https://wiki.freecad.org/Draft_SelectPlane
https://wiki.freecad.org/Draft_Snap

3+4: Duplicates.
#9457
#9458