FreeCAD/FreeCAD #15465: Cannot select a line on Sketch as the Custom Reference for a Mirror on PD Multitransform.

leoheck, 2024-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Cannot select a line on Sketch as the Custom Reference for a Mirror on Multitransform.
Also, this feature could be like the PD Pad, which shows a button and an Input field. 

[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git) AppImage
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-07-16:
Indeed, this would be useful.

FYI a way forward in the meantime is to attach a datum plane to that edge and mirror across that:

[image]

(Attached the plane normal to edge and flipped it around Y 90 degrees)

leoheck, 2024-07-16:
A minior improvment would also remove the applied mirrored operation from the list when a mirror is applied.

For instance, if I apply Vertical mirror, when I chose the mirror again, it should not select the Vertical Again (it is like that because it is the first item of the list). It could remove the vertical mirror from the list. Most of the time when using multitransform I mirror Vertically/Horizontally and then Horizontally/Vertically.

depthoffocus, 2024-07-16:
It’s not out of the question you’d want to mirror twice across the same axis at different points in a multitransform sequence. It’s not idempotent; the second mirror operation can be doing a different thing to the first.

maxwxyz, 2024-07-17:
@FlachyJoe FYI

leoheck, 2024-07-17:
@depthoffocus yeah, makes sense. how about changing the focus of the next mirror to the other option instead of removing it from the list?