FreeCAD/FreeCAD #15469: "WinError 998 Invalid access to memory location" when invoking "Part Design" Workbench

greggs123, 2024-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

"WinError 998 Invalid access to memory location" when invoking "Part Design" Workbench.
Seems to occur after working in other workbenches. Specifically Sheet Metal

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.14
  * Help 1.0.3
  * sheetmetal 0.4.20
[/code]
```

### Subproject(s) affected?

None

### Anything else?

No

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-16:
Could you try if the issue is also present in the current dev build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
TIA

greggs123, 2024-07-16:
Yes, I will install and play and then report back to you.

greggs123, 2024-07-18:
I have re-created the same part. So far have not seen the reported issue.
By the way, latest functionality is awesome. Can't wait for release.

maxwxyz, 2024-07-19:
Thanks, I'll close it then as fixed in next version.

closed 2024-07-19 by maxwxyz