FreeCAD/FreeCAD #15472: Measure:  Crash after click Menu > Tools > Measure without any open document

VM4Dim, 2024-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3f590) [0x7f8d86c56590]
#1  0x7f8d88be1ae0 in App::Document::getName() const from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libFreeCADApp.so+0
#2  0x7f8d66519cb1 in Gui::TaskMeasure::update() from /tmp/.mount_FreeCAGE5gyx/usr/lib/MeasureGui.so+0x511
#3  0x7f8d870e0830 in QObject::event(QEvent*) from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libQt5Core.so.5+0x260
#4  0x7f8d87f703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#5  0x7f8d89d3ad29 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libFreeCADGui.so+0x99
#6  0x7f8d870b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libQt5Core.so.5+0x108
#7  0x7f8d870b9801 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libQt5Core.so.5+0x181
#8  /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libQt5Core.so.5(+0x30b313) [0x7f8d8710b313]
#9  /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f8d85910ad7]
#10  /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f8d85913ac7]
#11  /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f8d859140ce]
#12  0x7f8d8710aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libQt5Core.so.5+0x66
#13  0x7f8d870b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libQt5Core.so.5+0x11b
#14  0x7f8d870bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libQt5Core.so.5+0x82
#15  0x7f8d89c9d4a2 in Gui::Application::runApplication() from /tmp/.mount_FreeCAGE5gyx/usr/bin/../lib/libFreeCADGui.so+0x972
#16  /tmp/.mount_FreeCAGE5gyx/usr/bin/freecad(+0x7620) [0x55d7813a9620]
#17  /lib/x86_64-linux-gnu/libc.so.6(+0x29c8a) [0x7f8d86c40c8a]
#18  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f8d86c40d45]
#19  /tmp/.mount_FreeCAGE5gyx/usr/bin/freecad(+0x7981) [0x55d7813a9981]
```

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git) AppImage
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-17:
It won't crash on Windows.
@hlorus FYI, maybe the command should be disabled if there is no document or active 3D view.

Syres916, 2024-07-17:
> It won't crash on Windows. @hlorus FYI, maybe the command should be disabled if there is no document or active 3D view.

Basically like it was for the old command https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Measure_Restrict_Availability

hlorus, 2024-07-17:
> > It won't crash on Windows. @hlorus FYI, maybe the command should be disabled if there is no document or active 3D view.
> 
> Basically like it was for the old command [main...Syres916:FreeCAD:Measure_Restrict_Availability](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Measure_Restrict_Availability)

Do you mind submitting your changes as a pull request?

Syres916, 2024-07-17:
> Do you mind submitting your changes as a pull request?

Done

PR #15481 (2024-07-21): [Measure] Correctly restrict command availability
Fixes #15472

FYI @hlorus
closed 2024-07-21 by chennes