FreeCAD/FreeCAD #15473: Leaving FreeCAD open for hours breaks rendering

flaviut, 2024-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I leave freecad open for a while (several hours in the background doing nothing), rendering breaks. I'm launching freecad using the appimage, with `QT_QPA_PLATFORM=xcb` set (since I'm on wayland and wayland support seems to be broken with the appimages). Note the missing icons in the screenshot in the tree panel & in the tasks panel, as well as the missing padding & incorrect fonts:

[image]

Here's what freecad typically looks like in this state:

[image]

I also see the following repeated in my console whenever I change which object I'm focused on:

```
QPainter::begin: Paint device returned engine == 0, type: 2
QPainter::setPen: Painter not active
QPainter::drawRects: Painter not active
QPainter::end: Painter not active, aborted
```

### Full version info

```shell
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git) AppImage
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

---

OS: Arch Linux (sway)
Word size of Ondsel: 64-bit
Version: 2024.2.2.37240 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.2)
Hash: fbb794cd2e56fba5fd911c8832332b5e5e235319
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

None

### Anything else?

I can also reproduce with Ondsel (and in fact I saw it there first)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-07-17:
@flaviut, opened, without doing anything? Does your computer turn on _Energy saving mode_?

How many messages have you received? Is it one-two or more than a few hundreds?

flaviut, 2024-07-17:
> opened, without doing anything?

Correct. The lock screen might automatically come on, but there's no interaction with the program.

> Does your computer turn on Energy saving mode?

I don't have any sort of special energy-saving modes. This is a desktop with a dedicated graphics card, I know some laptops have weirdness there.

However, I did find a sure-fire way to reproduce this: `swaymsg "output * power off"; sleep 30; swaymsg "output * power on"` to turn the video output off for 30s then turn it back on.

> How many messages have you received? Is it one-two or more than a few hundreds?

It's one set of messages per action taken. If I hide a body, I see a set of messages.

xtemp09, 2024-07-17:
Having stayed idle for 12 hours, _FreeCAD_ shows no such behavior. Turning monitor off and on using command-line tools shows the same result.

This problem seems to belong uniquely to _sway_.

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38239 (Git)
Build type: Debug
Branch: main
Hash: 0278bd8f27df454d34920e0f9a4243783b3e9ebf
Python 3.10.12, Qt 5.15.14, Coin 4.0.0, Vtk , OCC 7.5.1
Locale: English/United States (en_US)
```

flaviut, 2024-07-17:
Freecad must be run with `QT_QPA_PLATFORM=xcb` to reproduce this. Running on native wayland works just fine. From source, on the latest git, right clicking causes the following crash when in this state:

```
#0  0x00007fffee03955e in ??? () at /usr/lib/libQt5Widgets.so.5
#1  0x00007fffee02eefb in QFusionStyle::drawPrimitive(QStyle::PrimitiveElement, QStyleOption const*, QPainter*, QWidget const*) const () at /usr/lib/libQt5Widgets.so.5
#2  0x00007fffee034068 in QFusionStyle::drawControl(QStyle::ControlElement, QStyleOption const*, QPainter*, QWidget const*) const () at /usr/lib/libQt5Widgets.so.5
#3  0x00007fffee0e7a52 in QMenu::paintEvent(QPaintEvent*) () at /usr/lib/libQt5Widgets.so.5
#4  0x00007fffedf8f20a in QWidget::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#5  0x00007fffedf56331 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#6  0x00007ffff43aa832 in Gui::GUIApplication::notify(QObject*, QEvent*) () at /home/user/dev/FreeCAD/build/lib/libFreeCADGui.so
#7  0x00007fffed2aba68 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#8  0x00007fffedf8265e in QWidgetPrivate::sendPaintEvent(QRegion const&) () at /usr/lib/libQt5Widgets.so.5
#9  0x00007fffedf83cd8 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () at /usr/lib/libQt5Widgets.so.5
#10 0x00007fffedf62fd4 in ??? () at /usr/lib/libQt5Widgets.so.5
#11 0x00007fffedf7a0d3 in QWidgetPrivate::syncBackingStore(QRegion const&) () at /usr/lib/libQt5Widgets.so.5
#12 0x00007fffedfabb73 in ??? () at /usr/lib/libQt5Widgets.so.5
#13 0x00007fffedf56331 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#14 0x00007ffff43aa832 in Gui::GUIApplication::notify(QObject*, QEvent*) () at /home/user/dev/FreeCAD/build/lib/libFreeCADGui.so
#15 0x00007fffed2aba68 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#16 0x00007fffed73907b in QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) () at /usr/lib/libQt5Gui.so.5
#17 0x00007fffed71cdcd in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Gui.so.5
#18 0x00007fffe563f2f8 in xcbSourceDispatch (source=<optimized out>) at /usr/src/debug/qt5-base/kde-qtbase/src/plugins/platforms/xcb/qxcbeventdispatcher.cpp:105
#19 0x00007fffe9422ab9 in ??? () at /usr/lib/libglib-2.0.so.0
#20 0x00007fffe94849e7 in ??? () at /usr/lib/libglib-2.0.so.0
#21 0x00007fffe9421fc5 in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#22 0x00007fffed2fa37f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#23 0x00007fffed2a382c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#24 0x00007fffee0e7311 in ??? () at /usr/lib/libQt5Widgets.so.5
#25 0x00007fffee0e7436 in QMenu::exec(QPoint const&, QAction*) () at /usr/lib/libQt5Widgets.so.5
#26 0x00007ffff4a941c4 in Gui::TreeWidget::contextMenuEvent(QContextMenuEvent*) () at /home/user/dev/FreeCAD/build/lib/libFreeCADGui.so
#27 0x00007fffedf8fb2d in QWidget::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#28 0x00007fffee03af88 in QFrame::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#29 0x00007fffed2aaf02 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#30 0x00007fffedf56321 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#31 0x00007fffedf5c589 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#32 0x00007ffff43aa832 in Gui::GUIApplication::notify(QObject*, QEvent*) () at /home/user/dev/FreeCAD/build/lib/libFreeCADGui.so
#33 0x00007fffed2aba68 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#34 0x00007fffedfa9870 in ??? () at /usr/lib/libQt5Widgets.so.5
#35 0x00007fffedfaacc7 in ??? () at /usr/lib/libQt5Widgets.so.5
#36 0x00007fffedf56331 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#37 0x00007ffff43aa832 in Gui::GUIApplication::notify(QObject*, QEvent*) () at /home/user/dev/FreeCAD/build/lib/libFreeCADGui.so
#38 0x00007fffed2aba68 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#39 0x00007fffed73ba90 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/libQt5Gui.so.5
#40 0x00007fffed71cdcd in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Gui.so.5
#41 0x00007fffe563f2f8 in xcbSourceDispatch (source=<optimized out>) at /usr/src/debug/qt5-base/kde-qtbase/src/plugins/platforms/xcb/qxcbeventdispatcher.cpp:105
#42 0x00007fffe9422ab9 in ??? () at /usr/lib/libglib-2.0.so.0
#43 0x00007fffe94849e7 in ??? () at /usr/lib/libglib-2.0.so.0
#44 0x00007fffe9421fc5 in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#45 0x00007fffed2fa37f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#46 0x00007fffed2a382c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#47 0x00007fffed2afbfd in QCoreApplication::exec() () at /usr/lib/libQt5Core.so.5
[... cut ...]
#49 0x00007ffff3ff6804 in (anonymous namespace)::runEventLoop(Gui::GUISingleApplication&) () at /home/user/dev/FreeCAD/build/lib/libFreeCADGui.so
#50 0x00007ffff3ff6eca in Gui::Application::runApplication() () at /home/user/dev/FreeCAD/build/lib/libFreeCADGui.so
#51 0x0000555555582529 in main ()
```

```
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38259 (Git)
Build type: Unknown
Branch: main
Hash: c8046b1c60c786caca3050cf51f853454ffb5388
Python 3.12.4, Qt 5.15.14, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

xtemp09, 2024-07-18:
This is a known _Qt_ bug. [^1] [^2]

[^1]: https://bugreports.qt.io/browse/QTBUG-109331
[^2]: https://code.videolan.org/videolan/vlc/-/issues/27636

maxwxyz, 2024-07-18:
@xtemp09 is there anything how FreeCAD could work around that Qt bug?
Should this issue be closed then, supposed it'll work when the Qt bug is fixed?

xtemp09, 2024-07-18:
@maxwxyz, I don't know any workaround, sorry.

maxwxyz, 2024-07-18:
Will this issue be resolved automatically with a Qt upstream fix or is there more work needed after the updated Qt version?

xtemp09, 2024-07-18:
@maxwxyz, how would I know?

flaviut, 2024-07-18:
Looks like this isn't fixable by freecad

closed 2024-07-18 by flaviut