FreeCAD/FreeCAD #15474: ESC when chamfering/filleting causes crash

flaviut, 2024-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. create a pad of a rectangle in part design
2. click on an edge & select the chamfer/filet tool
3. with the default focus in the number field, type "1"
4. press the escape key. notice the crash

Stack trace (seems useless this time to be honest)

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3cae0) [0x766934a50ae0]
#1  [0x62f05a0617f0]
```

Can also be reproduced with ondsel, and the stack trace there may be more helpful:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3cae0) [0x7716e8250ae0]
#1  0x7716e7fe5a40 in QArrayData::shared_null from /tmp/.mount_ondselMzBlMh/usr/bin/../lib/libQt5Core.so.5+0
```

### Full version info

```shell
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git) AppImage
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

---

OS: Arch Linux (sway)
Word size of Ondsel: 64-bit
Version: 2024.2.2.37240 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.2)
Hash: fbb794cd2e56fba5fd911c8832332b5e5e235319
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-07-16:
Can confirm on:

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * Curves 0.6.40

Excepts from more voluminous Mac crash report details:

```
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000260
Exception Codes:       0x0000000000000001, 0x0000000000000260

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [57507]

[....]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   PartDesignGui.so              	       0x121b187c6 PartDesignGui::TaskDressUpParameters::setSelectionMode(PartDesignGui::TaskDressUpParameters::selectionModes) + 54
1   PartDesignGui.so              	       0x121b0c38a PartDesignGui::TaskChamferParameters::onSizeChanged(double) + 26
2   libQt5Core.5.15.13.dylib      	       0x107795d37 0x107649000 + 1363255
3   libFreeCADGui.dylib           	       0x1056e5b64 Gui::QuantitySpinBox::updateFromCache(bool, bool) + 292
4   libFreeCADGui.dylib           	       0x1056e37a7 Gui::QuantitySpinBox::validateInput() + 119
5   libFreeCADGui.dylib           	       0x1056e7148 Gui::QuantitySpinBox::focusOutEvent(QFocusEvent*) + 40
6   libQt5Widgets.5.15.13.dylib   	       0x106d13d1d QWidget::event(QEvent*) + 2003
7   libQt5Widgets.5.15.13.dylib   	       0x106df7b7c QAbstractSpinBox::event(QEvent*) + 152

```

xtemp09, 2024-07-17:
I get the following backtrace.

<details><summary>Backtrace</summary>

```cpp
Process 1702 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
    frame #0: 0x00007fffb7a0ba9c PartDesignGui.so`PartDesignGui::TaskDressUpParameters::getBase(this=0x0000555557cabed0) const at TaskDressUpParameters.cpp:411:36
   408      // was deleted. This exception will be likely unhandled inside the dialog and pass upper, But an error
   409      // message inside the report view is better than a SEGFAULT.
   410      // Generally this situation should be prevented in ViewProviderDressUp::setEdit()
-> 411      return pcDressUp->getBaseObject();
   412  }
   413 
   414  void TaskDressUpParameters::setSelectionMode(selectionModes mode)
(lldb) fr v
(const PartDesignGui::TaskDressUpParameters *const) this = 0x0000555557cabed0
(PartDesign::DressUp *) pcDressUp = nullptr
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
  * frame #0: 0x00007fffb7a0ba9c PartDesignGui.so`PartDesignGui::TaskDressUpParameters::getBase(this=0x0000555557cabed0) const at TaskDressUpParameters.cpp:411:36
    frame #1: 0x00007fffb7a0bb11 PartDesignGui.so`PartDesignGui::TaskDressUpParameters::showObject(this=0x0000555557cabed0) at TaskDressUpParameters.cpp:397:40
    frame #2: 0x00007fffb7a0bbb5 PartDesignGui.so`PartDesignGui::TaskDressUpParameters::setSelectionMode(this=0x0000555557cabed0, mode=none) at TaskDressUpParameters.cpp:420:19
    frame #3: 0x00007fffb7a033a1 PartDesignGui.so`PartDesignGui::TaskFilletParameters::onLengthChanged(this=0x0000555557cabed0, len=1) at TaskFilletParameters.cpp:145:21
    frame #4: 0x00007fffb7a04c6e PartDesignGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<double>, void, void (PartDesignGui::TaskFilletParameters::*)(double)>::call(f=<unavailable>, o=<unavailable>, arg=<unavailable>) at qobjectdefs_impl.h:152:20
    frame #5: 0x00007fffb7a04cac PartDesignGui.so`QtPrivate::QSlotObject<void (PartDesignGui::TaskFilletParameters::*)(double), QtPrivate::List<double>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) [inlined] void QtPrivate::FunctionPointer<void (PartDesignGui::TaskFilletParameters::*)(double)>::call<QtPrivate::List<double>, void>(arg=<unavailable>, o=<unavailable>, f=<unavailable>) at qobjectdefs_impl.h:185:95
    frame #6: 0x00007fffb7a04ca4 PartDesignGui.so`QtPrivate::QSlotObject<void (PartDesignGui::TaskFilletParameters::*)(double), QtPrivate::List<double>, void>::impl(which=<unavailable>, this_=<unavailable>, r=<unavailable>, a=<unavailable>, ret=<unavailable>) at qobjectdefs_impl.h:418:49
    frame #7: 0x00007ffff3ef819f libQt5Core.so.5`void doActivate<false>(QObject*, int, void**) [inlined] QtPrivate::QSlotObjectBase::call(a=0x00007fffffffb7f0, r=0x0000555557cabed0, this=0x0000555557c57ec0) at qobjectdefs_impl.h:398:57
    frame #8: 0x00007ffff3ef8183 libQt5Core.so.5`void doActivate<false>(sender=0x0000555558ec0950, signal_index=9, argv=0x00007fffffffb7f0) at qobject.cpp:3925:30
    frame #9: 0x00007ffff3eef16f libQt5Core.so.5`QMetaObject::activate(sender=<unavailable>, m=0x00000000019566e0, local_signal_index=1, argv=0x00007fffffffb7f0) at qobject.cpp:3985:26
    frame #10: 0x00007ffff7e02c1b libFreeCADGui.so`Gui::QuantitySpinBox::valueChanged(this=<unavailable>, _t1=<unavailable>) at moc_QuantitySpinBox.cpp:306:26
    frame #11: 0x00007ffff7e03c1e libFreeCADGui.so`Gui::QuantitySpinBox::updateFromCache(this=0x0000555558ec0950, notify=true, updateUnit=true) at QuantitySpinBox.cpp:587:32
    frame #12: 0x00007ffff7e03c9a libFreeCADGui.so`Gui::QuantitySpinBox::handlePendingEmit(this=0x0000555558ec0950, updateUnit=true) at QuantitySpinBox.cpp:569:20
    frame #13: 0x00007ffff7e05283 libFreeCADGui.so`Gui::QuantitySpinBox::validateInput(this=0x0000555558ec0950) at QuantitySpinBox.cpp:465:22
    frame #14: 0x00007ffff7e05da8 libFreeCADGui.so`Gui::QuantitySpinBox::focusOutEvent(this=0x0000555558ec0950, event=0x00007fffffffbe70) at QuantitySpinBox.cpp:904:18
    frame #15: 0x00007ffff4bbb444 libQt5Widgets.so.5`QWidget::event(this=0x0000555558ec0950, event=0x00007fffffffbe70) at qwidget.cpp:8793:22
    frame #16: 0x00007ffff4d597cb libQt5Widgets.so.5`QAbstractSpinBox::event(this=0x0000555558ec0950, event=0x00007fffffffbe70) at qabstractspinbox.cpp:822:26
    frame #17: 0x00007ffff7e05b8d libFreeCADGui.so`Gui::QuantitySpinBox::event(this=<unavailable>, event=<unavailable>) at QuantitySpinBox.cpp:878:35
    frame #18: 0x00007ffff4b6eee3 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(this=0x0000555555906060, receiver=0x0000555558ec0950, e=0x00007fffffffbe70) at qapplication.cpp:3640:31
    frame #19: 0x00007ffff4b77915 libQt5Widgets.so.5`QApplication::notify(this=0x00007fffffffd770, receiver=0x0000555558ec0950, e=0x00007fffffffbe70) at qapplication.cpp:3590:31
    frame #20: 0x00007ffff79e0cf2 libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00007fffffffd770, receiver=0x0000555558ec0950, event=0x00007fffffffbe70) at GuiApplication.cpp:83:40
    frame #21: 0x00007ffff3eb3cdd libQt5Core.so.5`QCoreApplication::notifyInternal2(receiver=0x0000555558ec0950, event=0x00007fffffffbe70) at qcoreapplication.cpp:1064:24
    frame #22: 0x00007ffff3eb3f5a libQt5Core.so.5`QCoreApplication::sendEvent(receiver=<unavailable>, event=<unavailable>) at qcoreapplication.cpp:1462:27
[... cut ...]
    frame #81: 0x00007ffff3429e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:107:1), argc=1, argv=0x00007fffffffdd48, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd38) at libc-start.c:392:3
    frame #82: 0x0000555555570275 FreeCAD`_start + 37
(lldb)

```

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38239 (Git)
Build type: Debug
Branch: main
Hash: 0278bd8f27df454d34920e0f9a4243783b3e9ebf
Python 3.10.12, Qt 5.15.14, Coin 4.0.0, Vtk , OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

maxwxyz, 2024-07-17:
Is this the same or related to: https://github.com/FreeCAD/FreeCAD/issues/15453 ?

FEA-eng, 2024-07-17:
As usual, no crash on Windows but I get those errors:

```
10:01:41  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

It doesn't happen in 0.21.2.

flaviut, 2024-07-17:
> Is this the same or related to: https://github.com/FreeCAD/FreeCAD/issues/15453 ?

May be related from the better stacktraces I'm seeing other folks get. It seemed like a separate issue from my stack trace.

wwmayer, 2024-07-19:
For further details about the crash: https://github.com/FreeCAD/FreeCAD/issues/15453#issuecomment-2238724516

closed 2024-07-22 by chennes