FreeCAD/FreeCAD #15478: CAM: sorting drill holes sorts alphabetically not by actual diameter (UI)

jffmichi, 2024-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sorting the holes in the Drilling operation task panel by diameter sorts alphabetically, i.e.  a dimater 38.4mm appears before a diameter of 4.5mm

[image]

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38174 (Git)
Build type: Release
Branch: main
Hash: a1487106ab745605f14b709a33f8ae9242d67584
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * kicadStepUpMod 11.1.7
  * dodo 1.0.1
  * sheetmetal 0.4.16
  * fcgear 1.0.0
  * fasteners 0.5.21
  * Assembly4 0.50.12
  * Curves 0.6.37
  * lattice2 1.0.0
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-17:
@jffmichi if you would like to try a change and let me know if it causes any regressions outputting the actual GCode it would be appreciated. At line 86 to 92 of _src/Mod/CAM/Path/Op/Gui/CircularHoleBase.py_ change on local build from:

```
                dia = obj.Proxy.holeDiameter(obj, base, sub)
                item = QtGui.QTableWidgetItem("{:.3f}".format(dia))
                item.setData(self.DataFeatureName, name)
                item.setData(self.DataObject, base)
                item.setData(self.DataObjectSub, sub)
                item.setTextAlignment(QtCore.Qt.AlignHCenter)
                self.form.baseList.setItem(self.form.baseList.rowCount() - 1, 1, item)
```
to:

```
                dia = obj.Proxy.holeDiameter(obj, base, sub)
                item = QtGui.QTableWidgetItem()
                item.setData(QtCore.Qt.DisplayRole, float(dia))
                item.setTextAlignment(QtCore.Qt.AlignHCenter)
                self.form.baseList.setItem(self.form.baseList.rowCount() - 1, 1, item)
```

Restart FreeCAD and re-test the form sorting, it appears to work correctly for me.

jffmichi, 2024-07-18:
That was fast. Your solution seems to work. Adding/removing holes for Drilling operation also still works and the correct holes are processed.

[image]

PR #15496 (2024-08-17): [CAM] Fix numeric sorting of drill diameters
Fixes #15478 which has been listed as a feature request, with relation to the use of toolchangers is a debatable point but I'll leave it to the Maintainers to decide if it should wait until Post 1.0
closed 2024-08-17 by sliptonic