FreeCAD/FreeCAD #15480: Preferences window changed to another one on language change

Chidorin, 2024-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I change Freecad language to russian Preferences(cmd+,) opens another window, not preferences.

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenDark 2023.12.17
```

### Subproject(s) affected?

None

### Anything else?

Freecad 0.22.0-38043

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-21:
@prokoudine do you mind trying to reproduce ?

maxwxyz, 2024-11-09:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12006

closed 2024-11-09 by maxwxyz