FreeCAD/FreeCAD #15484: TNP issue when replacing rectangular pad edge with more lines

FEA-eng, 2024-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Originally reported here: https://forum.freecad.org/viewtopic.php?t=87939

If you sketch a rectangle, add some zigzag lines replacing one of its edges but in construction mode first, then pad the rectangle, sketch two circles on it, go back to the first sketch and replace the edge with the zigzag lines by switching their modes then you will get a TNP error message for the second sketch.

I checked it in Link Branch and it doesn't happen there so perhaps it can still be fixed in main.

[TNP issue.zip](https://github.com/user-attachments/files/16273025/TNP.issue.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2024-07-23:
Confirmed,  Looking for it.

maxwxyz, 2024-07-29:
This should be fixed with #15575 
Please test again with the latest weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FEA-eng, 2024-07-30:
@bgbsww I thought that it was fixed by #15575 but I still have it in a fresh file. It just doesn't always occur. Interestingly, it doesn't seem to happen when the rectangle is of centered type but maybe there's another factor here.

```
09:57:10  <PropertyLinks> PropertyLinks.cpp(463): Unnamed#Sketch001.AttachmentSupport missing element reference Unnamed#Pad ;g1;SKT;:H2d9,E;FAC;:H2d9:4,F;:G0;XTR;:H2d9:8,F.Face6
09:57:10  PositionBySupport: AttachEngine3D: subshape not found Pad.?Face6
09:57:10  Sketch001: AttachEngine3D: subshape not found Pad.?Face6
09:57:10  Sketch001: AttachEngine3D: subshape not found Pad.?Face6
```

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

bgbsww, 2024-07-30:
Confirmed; modify the test for this to be based at 10,10 instead of 0,0 and it fails.

Interestingly, the face that isn't found seems to have an additional RFI  (refine) clause in the elementMap when it fails, so I suspect we are failing to be consistent on refine somewhere.

The last three faces ( Face6, Face7 and Face8 ) all have a refine after the switch out of the construction lines - those are the top, bottom, and one of the sides.

bgbsww, 2024-07-31:
I think I understand this now, and it might not be solvable:  Flipping on and off construction geometry does something fairly unique:  if the geometry being turned on or off is before other geometry used to create the face from the sketch that is then extruded by the pad, it breaks.  This is because it changes the history of the face to be based on different geom IDs, which changes the name, and then we can't look it up to perform the TNP mitigation.  So we punt and flag it as missing.  The basic rule here appears to be don't change the number of edges in the outer wire of a sketch, or we can't match the name.  There are a few cases where we get lucky, and little variations (like between LS3 and main) can affect where this happens.

So, I have an open conversation up to look for ideas, but this is not a simple bug - it is code correctly taking its fallback path when it can't guess for sure at the correct replacement.

bgbsww, 2024-08-01:
Okay, I don't think this is a blocker.  The solutions to this are
1) write smarter heuristics that successfully guess the right face - new code, out of scope right now
2) make a dialog to ask the user to tell you what to do - new code, out of scope and new UI not allowed

Here's an in depth example of a triangle sketch going to a rectangular one ( via construction geometry getting flipped ):
'Initial Sketch Element Map'
{'Edge1': 'g3;SKT',
 'Edge2': 'g4;SKT',
 'Edge3': 'g5;SKT',
 'Vertex1': ['g3v1;SKT', 'g5v2;SKT'],
 'Vertex2': ['g3v2;SKT', 'g4v1;SKT'],
 'Vertex3': ['g4v2;SKT', 'g5v1;SKT']}
'Initial Element Map'
{'Edge1': 'g3v1;SKT;:H166,V;:G;XTR;:H166:7,E',
 'Edge2': 'g3v2;SKT;:H166,V;:G;XTR;:H166:7,E',
 'Edge3': 'g3;SKT;:H166,E',
 'Edge4': 'g3;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E',
 'Edge5': 'g4v2;SKT;:H166,V;:G;XTR;:H166:7,E',
 'Edge6': 'g4;SKT;:H166,E',
 'Edge7': 'g4;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E',
 'Edge8': 'g5;SKT;:H166,E',
 'Edge9': 'g5;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E',
 'Face1': 'g3;SKT;:H166,E;:G;XTR;:H166:7,F',
 'Face2': 'g4;SKT;:H166,E;:G;XTR;:H166:7,F',
 'Face3': 'g5;SKT;:H166,E;:G;XTR;:H166:7,F',
 'Face4': 'g3;SKT;:H166,E;FAC;:H166:4,F',
 'Face5': 'g3;SKT;:H166,E;FAC;:H166:4,F;:G0;XTR;:H166:8,F',
 'Vertex1': ['g3v1;SKT;:H166,V', 'g5v2;SKT;:H166,V'],
 'Vertex2': 'g3;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E;:U;XTR;:H166:7,V',
 'Vertex3': ['g3v2;SKT;:H166,V', 'g4v1;SKT;:H166,V'],
 'Vertex4': 'g3;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E;:U2;XTR;:H166:8,V',
 'Vertex5': ['g4v2;SKT;:H166,V', 'g5v1;SKT;:H166,V'],
 'Vertex6': 'g4;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E;:U2;XTR;:H166:8,V'}
<PropertyLinks> PropertyLinks.cpp(461): SketchSolverTest#Sketch1.AttachmentSupport missing element reference SketchSolverTest#Pad ;g3;SKT;:H166,E;FAC;:H166:4,F;:G0;XTR;:H166:8,F.Face5
PositionBySupport: AttachEngine3D: subshape not found Pad.?Face5
Sketch1: AttachEngine3D: subshape not found Pad.?Face5
'Post Toggle Sketch Element Map'
{'Edge1': 'g1;SKT',
 'Edge2': 'g2;SKT',
 'Edge3': 'g3;SKT',
 'Edge4': 'g4;SKT',
 'Vertex1': ['g1v1;SKT', 'g4v2;SKT'],
 'Vertex2': ['g1v2;SKT', 'g2v1;SKT'],
 'Vertex3': ['g2v2;SKT', 'g3v1;SKT'],
 'Vertex4': ['g3v2;SKT', 'g4v1;SKT']}
'Post Toggle Element Map'
{'Edge1': 'g1v1;SKT;:H166,V;:G;XTR;:H166:7,E',
 'Edge10': 'g3;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E',
 'Edge11': 'g4;SKT;:H166,E',
 'Edge12': 'g4;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E',
 'Edge2': 'g1v2;SKT;:H166,V;:G;XTR;:H166:7,E',
 'Edge3': 'g1;SKT;:H166,E',
 'Edge4': 'g1;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E',
 'Edge5': 'g2v2;SKT;:H166,V;:G;XTR;:H166:7,E',
 'Edge6': 'g2;SKT;:H166,E',
 'Edge7': 'g2;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E',
 'Edge8': 'g3v2;SKT;:H166,V;:G;XTR;:H166:7,E',
 'Edge9': 'g3;SKT;:H166,E',
 'Face1': 'g1;SKT;:H166,E;:G;XTR;:H166:7,F',
 'Face2': 'g2;SKT;:H166,E;:G;XTR;:H166:7,F',
 'Face3': 'g3;SKT;:H166,E;:G;XTR;:H166:7,F',
 'Face4': 'g4;SKT;:H166,E;:G;XTR;:H166:7,F',
 'Face5': 'g1;SKT;:H166,E;FAC;:H166:4,F',
 'Face6': 'g1;SKT;:H166,E;FAC;:H166:4,F;:G0;XTR;:H166:8,F',
 'Vertex1': ['g1v1;SKT;:H166,V', 'g4v2;SKT;:H166,V'],
 'Vertex2': 'g1;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E;:U;XTR;:H166:7,V',
 'Vertex3': ['g1v2;SKT;:H166,V', 'g2v1;SKT;:H166,V'],
 'Vertex4': 'g1;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E;:U2;XTR;:H166:8,V',
 'Vertex5': ['g2v2;SKT;:H166,V', 'g3v1;SKT;:H166,V'],
 'Vertex6': 'g2;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E;:U2;XTR;:H166:8,V',
 'Vertex7': ['g3v2;SKT;:H166,V', 'g4v1;SKT;:H166,V'],
 'Vertex8': 'g3;SKT;:H166,E;:G;XTR;:H166:7,F;:U;XTR;:H166:7,E;:U2;XTR;:H166:8,V'}

Note:
 'Face5': 'g3;SKT;:H166,E;FAC;:H166:4,F;:G0;XTR;:H166:8,F',
 'Face6': 'g1;SKT;:H166,E;FAC;:H166:4,F;:G0;XTR;:H166:8,F',

You can see that we can't find the Face6 replacement for Face5, because the histories don't match, which is because we altered the geometry number ( edge) underlying it from being g3 to g1 - because it was a construction line toggle of geometry that already existed and had an assigned location.

I can imagine a heuristic that tried to match by plane and not by name potentially solving this; or maybe some clever renumbering of geometry in a sketch either on construction toggle or on exit and building the shape, but either one of those will cause confusing moments for users with things rename themselves.

So, @maxwxyz I think this shouldn't have a blocker tag, and I'm gonna unassign, at least for now

FEA-eng, 2024-08-02:
> Okay, I don't think this is a blocker. The solutions to this are
> 
> 1. write smarter heuristics that successfully guess the right face - new code, out of scope right now
> 2. make a dialog to ask the user to tell you what to do - new code, out of scope and new UI not allowed

I see. I just wonder if LS3 has special code to handle this that makes it work there or if it's a matter of some other differences.

bgbsww, 2024-08-02:
It fails under LS3 in lots of cases; try the triangular prism for starters.

FEA-eng, 2024-08-02:
> It fails under LS3 in lots of cases; try the triangular prism for starters.

Then I'm even more curious why it works there for the rectangle if there's no general code to handle such cases. But we may not be able to find out. Maybe it's a coincidence.

bgbsww, 2024-08-02:
Well, the exploration path for that would be to look carefully at the element maps in both cases and see what is going on with the names that causes this to work.

With that said, it occurs to me that the current question I'm looking at might actually be relevant:  main isn't hashing element names, and LS3 is.  The hashing element squashes the text out and just leaves the length, which would allow examples like the one above that differ by one character near the start of the name to work just fine - it would match.

The tests for this would be to turn off hashing in LS3 - "App.ActiveDocument.UseHasher = False" and see if the problem happens, or to check in main after it comes on correctly.  I'll look at that second part after it works correctly.

FEA-eng, 2024-08-02:
> With that said, it occurs to me that the current question I'm looking at might actually be relevant: main isn't hashing element names, and LS3 is. The hashing element squashes the text out and just [...]

That's interesting. Is there a plan to add this hashing to main in the next release cycle? Would it be difficult to implement?

bgbsww, 2024-08-02:
It is a bug that it isn't happening now.  The code is there, just trying to
find why.  Hopefully later today it will be on.

bgbsww, 2024-08-03:
I think this is fixed, you need to do a recompute as the first step to update the element map.

bgbsww, 2024-08-03:
This raises an interesting  point, which is that having the hasher off will always cause this class of bugs to resurface.  The hashing of the names eliminates enough information that elements that otherwise would not match do.  Since this is likely  to only be used while debugging it won't affect users, but might bite a developer.

FEA-eng, 2024-08-06:
@bgbsww Unfortunately, merging #15734 didn't help, I still have this issue in the newest weekly just like I had it when testing the PR. Maybe there are still some differences between your build and master.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

bgbsww, 2024-08-08:
A couple more points here:

@maxwxyz this is not a blocker.  It cannot be fixed and it can be duplicated in LS3.

If you take the example file at the top and try it in LS3 it fails there too in exactly the same way.

Basically it depends on how you draw the rectangle and the zig-zags - starting with a different corner to draw the rectangle will change your results - thus my thinking this was fixed inadvertently with the hashing; nope, just a
different drawing style.

With that same file toggle the zigzags on and the 3 edges to the right to construction to form a small zigzag shape, you'll discover that the error doesn't happen.  That's because you aren't altering Edge1 in that case, and the top face then behaves.

Some other stuff:   Make a rectangle sketch and pad it and attach something to the top.  Then open that sketch, delete the bottom edge and replace it with two lines:  works just great when you close the sketch.  Try the same thing, except this time draw the two lines first, and then delete the bottom edge:  close the sketch and the attachment fails with a missing edge.  Again, both in main and LS3, exactly the same way.

Ultimately this is because of two things:  The sketch outer wire mapping of geometry elements is fragile, especially around edge1, and because we don't support any heuristics other than the element history for guessing what the correct replacement is after an attachment breaks.  The latter can definitely be fixed post 1.0 but not before.

The former has had multiple sets of eyeballs looking at it for quite a few hours, and nothing is solving it.

sslupsky, 2024-09-19:
Wondering if an AI model could be trained to assist the mapping?

sslupsky, 2024-09-19:
I noticed that the taper example provided in this forum post:
https://forum.freecad.org/viewtopic.php?p=782806#p782806

causes an error when the Map Mode is TangentPlane but there is no error when the Map Mode is FlatFace.