FreeCAD/FreeCAD #15485: TNP: Faces not following sketch/boolean cut tool rotation

FEA-eng, 2024-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Originally reported here (3 versions - Part, Part Design and Part Design with sketch instead of boolean): https://forum.freecad.org/viewtopic.php?t=88796#p768458

1. Note the face colors:

[image]

2. Rotate the "turn_axis_90degree" object by 90 degrees using its Placement angle. The faces should swap but it doesn't happen and they stay there.

It doesn't seem to happen in Link Branch so I'm reporting it here in case it can still be fixed for 1.0.

[TNP_face_colors.zip](https://github.com/user-attachments/files/16273382/TNP_face_colors.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2024-07-24:
I'm at a real loss here for why this labelled as a TNP error.  It wouldn't surprise me if there are behavior differences with LS3. but changing a placement value doesn't change the geometry, so it doesn't change the element map, so there's no possible TNP here, unless I'm missing something.

Checking the Body001.Shape.ElementMap before and after the placment angle change confirms this.

FEA-eng, 2024-07-24:
@bgbsww What I shared here is one of the three variants of this issue from the forum thread. The other two are done in Part Design - the last one with placement of the pocket sketch being changed. Is it also out of the scope of TNP?

bgbsww, 2024-07-24:
No, I think the others may still be in play.  This as described and in isolation does not appear to be TNP related at all.  The question of "guessing" the correct element after a detected failure is definitely in play.  And there's a massive difference in the code that I'm still escorting through PRs that may address some of the other issues - at the very least it will need testing then.

So this is a narrowly defined not-this-one.

bgbsww, 2024-08-11:
@maxwxyz As stated, this does not appear to be a TNP bug, so Topic::Toponaming isn't appropriate ( and the description is wrong)

chennes, 2024-08-31:
I propose dropping this from the 1.0 blockers list. What do you think, @maxwxyz?

maxwxyz, 2024-08-31:
Fine with this, the other example from the forum thread also is an issue in RT's branch

wsteffe, 2025-10-08:
When you turn body named turn_axus_90degree the faces of turn_axus_90degree/Pad001 swap as expected. 
Why do you expect that also faces of Cut (shown in picture) must swap ?