FreeCAD/FreeCAD #15486: TNP: Error when changing previous Pad type from To first to Dimension

FEA-eng, 2024-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Originally reported here: https://forum.freecad.org/viewtopic.php?t=87966#p761944

[TNP_pad_type.zip](https://github.com/user-attachments/files/16273835/TNP_pad_type.zip)

Just switch Pad001 from To first to Dimension.

It happens in Link Branch as well but might be worth reporting anyway.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-18:
This looks like a duplicate of #14595 which has already been decide to drop the Blocker label and just keep as a Bug while trying for 1.0. The working file and steps to reproduce have been provided see https://github.com/FreeCAD/FreeCAD/issues/14595#issuecomment-2213697652

FEA-eng, 2024-07-18:
@Syres916 The model is very similar but there's a different situation happening in both files so I prefer to keep the issue until it's confirmed to have the same underlying cause. Also, someone should make sure that it also doesn't work in the newest Link Branch version available at the moment. I can try to do it but I'm not sure where I can find such a version for Windows (or Ubuntu VM).

Syres916, 2024-07-18:
OK, I have the latest _LinkStable_ which realthunder updated yesterday, self-compiled so I can recheck. Just for info, to clone a specific branch using LinkStable as an example:

`git clone -b LinkStable --single-branch https://github.com/realthunder/FreeCAD/FreeCAD.git freecad-LinkStable-source`

FEA-eng, 2024-07-18:
@Syres916 So it seems that Link Daily was replaced by Link Stable ? A bit misleading. Anyway, I can try compiling it too if it's not very different from FreeCAD 0.22 compilation.

Syres916, 2024-07-18:
Before we go down a rabbit hole, could you please test the following file with FC main build 38259 or later. I think we've got more of an issue with bringing in older files then TNP analysis, just remove the `.zip` extension. Obviously the user can't bring the dimension lower than 7mm using standard configs.

[topo_t1_38259.FCStd.zip](https://github.com/user-attachments/files/16282891/topo_t1_38259.FCStd.zip)

FEA-eng, 2024-07-19:
> Before we go down a rabbit hole, could you please test the following file with FC main build 38259 or later.

The current weekly build is still 38231 so I built from the source myself. I see those message in the report view after opening the file shared by you:

```
12:41:25  Reading failed from embedded file: StringHasher.Table.txt (198 bytes, 97 bytes compressed)
12:41:25  <ElementMap> <string>(1): No hasherRef
12:41:25  <ElementMap> <string>(1): No hasherRef
```

When I extend the middle pad, as you did in #14595, I get:

```
12:45:33  <PropertyLinks> PropertyLinks.cpp(452): topo_t1_38259#Sketch002.AttachmentSupport missing element reference topo_t1_38259#Pad001 ;g17;SKT;:H135c,E;:G;XTR;:H135c:7,F;:H-1361,F.Face12
12:45:33  PositionBySupport: AttachEngine3D: null subshape
12:45:33  Sketch002: AttachEngine3D: null subshape
12:45:33  Sketch002: AttachEngine3D: null subshape
```

Is that what you wanted me to test?

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38260 (Git)
Build type: Unknown
Branch: main
Hash: ff61f56d19f65dcb9b8cb4b4768d54c28346c359
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.5.1
Locale: English/United States (en_US)
```

Syres916, 2024-07-19:
Thanks for the feedback, yep you did the correct test, the `Reading failed from embedded file: Stringhasher.Table.txt` is currently expected in _some_ circumstances and I've already submitted a PR to improve user feedback for Post 1.0 see https://github.com/FreeCAD/FreeCAD/pull/15229

So the marking of the whole document for Recompute and subsequent Refresh is critical in this instance. I must admit as I saved the file after a Recompute I didn't expect the `Stringhasher.Table.txt` error in this case, shows how little I know how this hangs together but you now start to understand why realthunder has a _Recompute?_ messagebox on opening his files.
I'm still waiting on recompiling to the same commit but I'm anticipating the recompute will sort the AttachEngine3D errors.

Syres916, 2024-07-19:
As expected, once the whole document has been recomputed it works as I would expect. My simple logic is the root cause lies with the saving of the element map, something is missing on save and so on restoring the incomplete element map the Pad001 doesn't have a clue where Sketch002 is attached and so throws the `AttachEngine3D: null subshape` errors.

As there's now several PRs with Topo related code being reviewed, I'd say park this for the time being, let those additional code merges take place

I'll get the Unit Test https://github.com/CalligaroV/FreeCAD/compare/toponaming-Part-makeElementWires-from-LS3...Syres916:FreeCAD:patch-4 perfected so it'll be ready for the beginning of next week as well as one that opens a saved file to show the difference.

I've got a raft of CAM tests to do alongside it and they are really tedious if I'm being honest so it's just a case of breaking up my day so I don't go cross eyed!!

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38260 (Git)
Build type: Release
Branch: main
Hash: ff61f56d19f65dcb9b8cb4b4768d54c28346c359
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * FreeCAD_SketchArch
  * FreeCAD-themes 2024.7.24
  * sheetmetal 0.4.23
```

FEA-eng, 2024-07-19:
I wanted to test the issue reported here with Link Branch but my own build is not working correctly and the newest Link Stable release (from April) fails to finish pad properly in the To first mode. No idea why it doesn't work there. Can you check this one in Link?

> As there's now several PRs with Topo related code being reviewed, I'd say park this for the time being, let those additional code merges take place

It would be good if @bgbsww could take a look at this. Maybe it hasn't been considered yet and the current PRs cover completely different stuff. But two of my recent TNP bug reports are confirmed to work in Link: #15484 and #15485 so I guess they should have priority.

Syres916, 2024-07-19:
OK, two different results, LinkStable is a no go looks as though it's not happy with the file, it opens but using:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.39003 (Git)
Build type: Release
Branch: LinkStable
Hash: 544eb3c4fd80a3ab699e76e4cebe1a5bc7b68d84
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * FreeCAD_SketchArch
  * FreeCAD-themes 2024.7.24
  * sheetmetal 0.4.23
```

All Pads are removed and trying to Pad for any sketch results in:

```
<Exception> Cannot add static property
<Exception> Cannot add static property

```

but LinkDaily :

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2023.1231.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: 7471b6a2720ee4973b9477a55713070b056c42e8
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * FreeCAD_SketchArch
  * FreeCAD-themes 2024.7.24
  * sheetmetal 0.4.23
```

All works the same as our main build 38259 or later

bgbsww, 2024-07-19:
Oh, I'm watching.  And working to get the save/load all correct.  It is
worth mentioning that LS3 intentionally does not save element maps in order
to force a recompute on reload [not sure is it is all cases].  You can see
this if you unzip and examine the FCStd files.  Or as a proxy, in the
.Contents output (useful for tests).  This and the two referenced are on my
radar, but it makes more sense to get save/load to a "believed to have no
errors" state and then retest these and hunt them down if necessary.

FEA-eng, 2024-07-19:
Correct me if I'm wrong but I'm starting to think that it's best to recreate the models if they are so simple when testing TNP issues in main or LS3. I'm not sure if opening those files and doing the changes is enough for the TNP mitigation code to work.

bgbsww, 2024-07-19:
If what you are questioning is the validity of files attached to bug
reports that come from the bad code, then yes, those currently need to be
approached with a fair degree of caution, recalculated certainly, and
sometimes recreated.  That is why save/load is the top priority.  Note that
the scope appears to be sketches, and at the most detailed level external
references in sketches.  Errors in operations between primitives that are
not based on sketches haven't thrown up a bug in a while.  Unfortunately,
this is woven into sketcher and it's dependencies, which is why it is
taking a while to fix.  The gold standard here is a test, second best is a
macro, third is detailed instructions to reproduce, and fourth is "this
file is broken".  Video captures are useful when the focus is on steps to
reproduce as opposed to the final click that causes the fail.

bgbsww, 2024-07-29:
Can this be retested now that #15575 is in place?  It should be in tomorrow's weekly.

FEA-eng, 2024-07-30:
> Can this be retested now that #15575 is in place? It should be in tomorrow's weekly.

Unfortunately, it's not fixed yet. I still get errors here (after recreating the file):

```
09:37:50  PositionBySupport: AttachEngine3D: subshape not found Pad001.?Face12
09:37:53  Sketch002: AttachEngine3D: subshape not found Pad001.?Face12
```

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

bgbsww, 2024-08-06:
I can clearly see a bug specifically with the Dimension type - two Dimensions works fine.

sslupsky, 2024-09-19:
> Digging all the way down on this, the element lookup is failing because after you change from UpToFirst ( or any Up To mode to a Length mode, you change algorithms, fewer steps are done, meaning that the history of the faces is completely different due to different operations used to get there. I can't see any way for this to have worked in any version, but if you can point to one I'll take a look.

This forum thread might be related:
https://forum.freecad.org/viewtopic.php?p=782806#p782806

drwho495, 2026-09-05:
@maxwxyz could you add "fixed by V2"?

luzpaz, 2026-10-08:
@maxwxyz most likely Toponaming v2.0 fix won't make it into 26.3, right ? Should these bugs be moved 27.1 in that case ?

drwho495, 2026-10-08:
> 27\. [@maxwxyz](https://github.com/maxwxyz) most likely Toponaming v2.0 fix won't make it into 26.3, right ? Should these bugs be moved 27.1 in that case ?

Yeah, that would be for the best

maxwxyz, 2026-10-09:
@luzpaz I'll move related PRs and issues after the release. Please bellt move regressions as the milestone indicates which version broke it