FreeCAD/FreeCAD #15487: TNP issue when changing the tip

FEA-eng, 2024-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Originally reported here: https://forum.freecad.org/viewtopic.php?t=88796#p768460

[TNP_issue_tip.zip](https://github.com/user-attachments/files/16273935/TNP_issue_tip.zip)

Just set the tip to Revolution and make a groove with Sketch001 to get TNP errors.

It happens for me with Link Stable as well but I haven't tested with the newest version of Realthunder's fork.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-18:
In this case, the both LinkDaily and LinkStable generate the same error as FC Build 38259 but (and it's a big but) when the user double-clicks the _Fillet_ the Link builds give a guess (and in my test the correct guess of Edge9) whereas the FC Build 38259 does not suggest the guess. Here's the outputs and build references:

```
<Exception> FeatureDressUp.cpp(140): Invalid edge link: ?Edge5
Recompute failed!
<PartDesignGui> Utils.cpp(1725): missing element reference in xwgvlkoe#Fillet.Base: ;#2f;:G;RVL;:H91b:7,E.Edge5
 <PartDesignGui> Utils.cpp(1736): guess element reference in xwgvlkoe#Fillet.Base: ;#2f;:G;RVL;:H91b:7,E.Edge5 -> #36;:H91f,E

```

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2023.1231.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: 7471b6a2720ee4973b9477a55713070b056c42e8
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * FreeCAD_SketchArch
  * FreeCAD-themes 2024.7.24
  * sheetmetal 0.4.23
```

```
<Exception> FeatureDressUp.cpp(140): Invalid edge link: ?Edge5
Recompute failed!
<PartDesignGui> Utils.cpp(1727): missing element reference in xwgvlkoe#Fillet.Base: ;#1b;:G;RVL;:H91b:7,E.Edge5
<PartDesignGui> Utils.cpp(1738): guess element reference in xwgvlkoe#Fillet.Base: ;#1b;:G;RVL;:H91b:7,E.Edge5 -> #23;:H91c,E
```

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.38998 (Git)
Build type: Release
Branch: LinkStable
Hash: 0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * FreeCAD_SketchArch
  * FreeCAD-themes 2024.7.24
  * sheetmetal 0.4.23
```

```
<Exception> FeatureDressUp.cpp(209): Invalid edge link: ;Edge5.Edge5
Fillet: Invalid edge link: ;Edge5.Edge5
```

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38259 (Git)
Build type: Release
Branch: main
Hash: c8046b1c60c786caca3050cf51f853454ffb5388
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * FreeCAD_SketchArch
  * FreeCAD-themes 2024.7.24
  * sheetmetal 0.4.23
```

bgbsww, 2024-07-24:
Thanks for all the detail here.  At first read, this is a run screaming kind of thing, because there's a whole new property file PropertyDressUp.cpp defining PropertyFilletSegments and PropertyChamferEdges which are then used, along with new property definitions for the FeatureFillet and FeatureChamfer functions.  There's changes to to saved file formats as well.  So much more that a quick bug fix; it's a big set of new capabilities to be carefully vetted.

However,, as it turns out there's a little hunk of code in PartDesign/Gui/Utils.cpp that seems to be doing this guessing that you identified in you log above.  I'm not yet sure I can disentangle it from the 1200 lines of stuff added into that file, but it does look very promising.  PartDesignGui::populateGeometryReferences is called exactly once from TaskDressUpParameters::populate as called once by TaskDressUpParameters::refresh from chamfer, fillet thickness and draft 's versions of ::refresh().

Unfortunately, this version of ::refresh() doesn't look like it's commonly used in main, so there are two steps here:

1)  Determine if refresh() gets called if defined and is an appropriate mechanism, or identify a good substitute.
2)  See if the guessing code comes across cleanly.

If those are both affirmative, then this has potential for 1.0.  If either isn't, then improving the guessing code might need to be a post 1.0 thing.

bgbsww, 2024-07-27:
Separate from the question about guessing the value here, it appears that the file itself has a quesitonable entry in it:
                <Property name="Base" type="App::PropertyLinkSub">
```
                    <LinkSub value="Revolution" count="1">
                        <Sub value="Edge5" shadow=";Edge5.Edge5"/>
                    </LinkSub>
                </Property>

```  ;Edge5.Edge5 doesn't look legit to me, so I want to figure out how it got there and if that bug has been fixed.  It's inside the Revolution.

maxwxyz, 2024-07-29:
This should be fixed with #15575 
Please test again with the latest weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FEA-eng, 2024-07-30:
Rechecked with the newest weekly. The errors still occur:

```
10:40:16  <PropertyLinks> PropertyLinks.cpp(463): Unnamed#Fillet.Base missing element reference Unnamed#Groove ;Edge5.Edge5
10:40:16  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;Edge5.Edge5
10:40:16  Fillet: Invalid edge link: ;Edge5.Edge5
```

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

bgbsww, 2024-08-01:
I've found the code here that "guesses" the correct edge, but it is heavily intertwined with completely different UI code with different dialogs.

I've factored out the "guess" part to add to Part::Feature, and I'm almost there on making minor changes to the existing dialog code to call it.  So this should get fixed shortly.

FEA-eng, 2024-08-06:
Unfortunately, I'm still getting TNP errors from this (recreated model) in the newest weekly so the issue may need to be reopened.

```
09:29:48  <PropertyLinks> PropertyLinks.cpp(463): Unnamed#Fillet.Base missing element reference Unnamed#Groove ;Edge5.Edge5
09:29:48  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;Edge5.Edge5
09:29:48  Fillet: Invalid edge link: ;Edge5.Edge5
```

@maxwxyz Can you reopen?

@bgbsww If those tests are working for you, maybe something else from your build is still missing in master.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

maxwxyz, 2024-08-06:
@FEA-eng could you start recording a macro before recreating it from scratch and uploading it here?

FEA-eng, 2024-08-06:
> @FEA-eng could you start recording a macro before recreating it from scratch and uploading it here?

Thanks. That's a very good idea, I don't why I haven't thought about that. However, the macros that I record can't be used to recreate the model for some reason. Those seem to be quite nasty limitations/bugs of the macro recording functionality. I will need to consult it on the forum.

maxwxyz, 2024-08-06:
Nonetheless @bgbsww could better see what is going on

bgbsww, 2024-08-06:
Download attached file.
Open
Mark entire file for recompute
ctrl-r
right click revolution, set tip
select Sketch001
hit groove:   Invalid ;Edge5.Edge5 as expected.  Fillet icon goes invalid
Double click fillet to open dialog:   Edge9  (correct guess) is shown.
Hit ok

Everything is as should be.  Does this not happen for you?

FEA-eng, 2024-08-06:
> Double click fillet to open dialog: Edge9 (correct guess) is shown.

Ah, that's the trick. I thought that it would eliminate the error completely but it works differently - allows to correct the issue automatically once it occurs. Sorry for the confusion, then this seems to be solved and since you said that there are no plans to prevent the error from occurring in the first place then I'm closing the issue. Thanks again.

closed 2024-08-06 by FEA-eng

bgbsww, 2024-08-06:
Thanks, one note:  there is thinking and a little experimentation towards popping up a dialog at the instant we discover a missing reference, but that has to be post 1.0 and is going to take both a POC and a discussion, since it is both technically and user experience tricky.

PR #15740 (2024-08-05): Toponaming: add code to implement guessing missing references
Fix #15487