FreeCAD/FreeCAD #15488: Sketcher no longer respecting "Ask for value after creating dimensional constraint"

theosib, 2024-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have always had "Ask for value after creating dimensional constraint" set, and it still is set. But in 38231, this setting is no longer being respected. So if I, say, select two points and press "L" or "I" or "KD" to set a distance constraint, it just sets the constraint to the current value rather than asking for one. This is a regression since the last dev release I tried, which is 38079.

[image]

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-07-17:
I talked to people on Discord about this. It's working okay for them, but it's 100% reproducible for me. I also tried turning the setting off and back on again. No change. One person pondered if this might be related to the locale bug.

krushia, 2024-07-17:
I just checked with C locale and it still works here, so that can be ruled out.

Also tried with/without "On view parameters" and works either way.

theosib, 2024-07-18:
The previous snapshot I have installed is 38097. The bug does NOT manifest in that version.

Where can I find the weekly builds between 38097 and 38231? I can narrow down when the bug was introduced.

theosib, 2024-07-18:
Well, after loading 38097 and then loading 38231, the problem went away. Previously I had been using 0.21.2 a bunch since PartDesign screw threads don't work in the dev builds. I'm betting that if I messed around loading different designs into different versions, it would mess up something in a config file or the active state and make the bug manifest again.

maxwxyz, 2024-07-18:
Might be because of the config files. I'll close this then. Feel free to reopen if there is a reliable way how the bug is reproducible.

closed 2024-07-18 by maxwxyz