FreeCAD/FreeCAD #15489: Cannot assign Backspace key on Mac to Std_Delete

theosib, 2024-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Please see https://github.com/FreeCAD/FreeCAD/issues/15069 for reference. There someone else complained about the Delete (actually Backspace) key on Mac not working to delete things. We can argue about whether or not this is a bug (I think it is), but I'm trying to follow the instructions given there to reassign the Backspace key to the Std_Delete function.

So, here we have the Customize dialog, where I have searched for Delete.
[image]

Currently, it has Delete selected. So I click in the "New shortcut" field and press Backspace. The Backspace symbol appears.
[image]

Then I press Assign.

However, it doesn't take. Instead a ? appears.
[image]

In this state, I have tried deleting something (a line segment in Sketcher), and the Backspace key doesn't do anything. However, if I press Fn+Backspace, which is the default, it still works.

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-07-17:
I also tried tabbing out of the "New shortcut" field before pressing Assign, but that didn't help.

theosib, 2024-07-17:
BTW, I can assign something stupid like the letter T, and it'll accept it. Just not the Backspace key.

luzpaz, 2024-07-17:
Duplicate of #9639  
Closing

closed 2024-07-17 by luzpaz