FreeCAD/FreeCAD #15491: Start Page greedy focus after creating project

obelisk79, 2024-07-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After freshly launching FreeCAD. The Start Page will greedily retake focus after the following actions:

1. Create New File from the template buttons
2. Open Recent File
3. Open Example File

It does *not* take focus if creating new file from the application menu.

This is only present after initial launch of the software. To retest, you have to close and relaunch.

### Full version info

```shell
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38243 (Git)
Build type: Release
Branch: main
Hash: 44387a793b062c60c8864ac564a9b2bcb3210fce
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-18:
@chennes FYI

krushia, 2024-07-18:
Maybe related to https://github.com/FreeCAD/FreeCAD/issues/12266 ?

Syres916, 2024-07-19:
I can't replicate this even using the same distro. Clean config files, launch FC and don't click anything else, just the _PartDesignExample.FCStd_ file and focus remains on the file.

I have also tried using a reasonably thorough config with overlay enabled with theme config from https://github.com/Syres916/FreeCAD-themes/tree/Behave_dark_TechDraw_Overlay_Start_fixes/Behave-dark and it still works as expected though it would take me a bit to get used to the main window appearing to momentarily close and re-open when clicking on a file but that's no doubt due to the bleeding edge libraries.

```
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38259 (Git)
Build type: Release
Branch: main
Hash: c8046b1c60c786caca3050cf51f853454ffb5388
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
```

krushia, 2024-07-24:
This happened to me before but it stopped. I've come across 3 reports of this and they all had OpenDark installed (as did I at some point). I'm not blaming OpenDark, but based on the depreciation date I suspect we all had configurations that bridged the removal of the Start workbench.

Could it be workbench auto-load settings?

Syres916, 2024-07-25:
As I never mix config files from different versions, I have a set of 0.21 config files which I'll test after installing Open themes on the EndeavorOS vm. Hopefully then the bug trace can begin.

Syres916, 2024-07-28:
Well, I installed OpenThemes copied over my 0.21 config files, invoked OpenDark and restarted FreeCAD but it still worked as expected, sorry I couldn't be any help but I'll have to bow out of this now.

obelisk79, 2024-07-28:
This is not an issue related to OpenThemes. I just did a fresh install, cleared my mods directory, fresh config. 

Set theme to FreeCAD Light, set overlay mode, separate task view and dock it on the right, leave tree and property panel on the left. Create new file using the links from start page. Everything works as should. Restart FreeCAD, change no additional settings, and I get the greedy focus by the start page again.

obelisk79, 2024-07-28:
Works fine if overlay is disabled. This issue didn't present itself until the new start page was implemented.

maxwxyz, 2024-08-08:
I can confirm this on macOS in the latest weekly using FreeCAD Classic theme.
Start page is open and I open a new page.
Each time I want to change a property in the property editor and hit enter to confirm, I am taken to the start page.
I hope you're ok if I'm labeling this as a blocker as this is a very annoying behavior.

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```
I cannot reproduce this on Windows.

maxwxyz, 2024-08-08:
Here is a demo:
[image]

chennes, 2024-08-14:
Can you test closing the start page and see if instead you get switched back to the first available tab every time?

maxwxyz, 2024-08-14:
@chennes yes, that is the case. Is it also the same at your side @obelisk79 ?
If so, I'd close this as duplicate of https://github.com/FreeCAD/FreeCAD/issues/14350

xtemp09, 2024-08-15:
Is there a way to reproduce it on _KUbuntu_? No matter how I tried, _FreeCAD_ works as expected on _KUbuntu 22.04.4_.

Issues #14350 and #12266 are not reproducible as well.

chennes, 2024-08-27:
Noting for posterity that this is NOT a duplicate of https://github.com/FreeCAD/FreeCAD/issues/14350 (and @maxwxyz's confirmation above is really a confirmation of that issue). So far only @obelisk79 has been able to reproduce, and my understanding is that is true only for one system.

obelisk79, 2024-08-27:
It is odd. Also of note I recently reinstalled my system fresh and the behavior remains. Perhaps it is my unique configuration of hardware.

maxwxyz, 2024-11-03:
@obelisk79 is this still an issue, also with the RC versions?

obelisk79, 2024-11-03:
No it no longer seems to be an issue.

closed 2024-11-03 by obelisk79