FreeCAD/FreeCAD #15494: "No autoconstraints added, please report" message received.

shadowbane1000, 2024-07-18:
[panels.zip](https://github.com/user-attachments/files/16229570/panels.zip)

### Bug:
When creating a slot with autoconstraints on, in the particular situation of the sketch above, the following message is received:
20:07:29  Autoconstraints cause redundancy. Removing them
20:07:29  Redundant constraint is not an autoconstraint. No autoconstraints or additional constraints were added. Please report!
And the slot does not appear in the view.

### Expected result:
The slot should have been created with autoconstraints.

### Steps to reproduce:

1. Using FreeCAD_weekly-builds-38127-conda-Windows-x86_64-py311
2. Open the attached project file.
3. Edit Sketch001
4. In the bottom right corner of the panel find a reference slot feature [image]
5. delete the two arcs and two lines that make up the feature.
6. Make sure auto constraints is turned on.
7. recreate the slot by clicking first on the center of the left circle, then the center of the right circle, then somewhere along the edge of about where the slot belongs.
8. The error should now be displayed.

luzpaz, 2024-07-24:
@shadowbane1000 please add your full [About](https://wiki.freecad.org/About) info. TIA

shadowbane1000, 2024-07-24:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git)
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.15
  * freecad.gears 1.0.0

FEA-eng, 2024-07-29:
Then the slot appears if you draw anything else or exit the sketch. It seems to be the same as what I've encountered when testing #15549:

[image]

maxwxyz, 2024-07-29:
So this then should also be fixed when #15549 is fixed (with https://github.com/FreeCAD/FreeCAD/pull/15575)
So please test again with the latest weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FEA-eng, 2024-07-30:
> So this then should also be fixed when #15549 is fixed (with #15575) So please test again with the latest weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Unfortunately, I still have this error and weird behavior with disappearing slot in my simplified case based on your report about crash. Same in the file attached to this issue but maybe such models should be recreated in each new dev version.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

maxwxyz, 2024-07-30:
@shadowbane1000 could you report back, also with the issue created from scratch?

maxwxyz, 2024-08-13:
any news on this?
@PaddleStroke FYI

maxwxyz, 2024-11-18:
closing for no further feedback