FreeCAD/FreeCAD #15503: Material: Child ignoring parent material

davesrocketshop, 2024-07-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When modifying a part, the resulting child ignores the parent material.

Steps to reproduce:
1. In Part WB create a Box
2. Set the material to something other than default
3. Create a chamfer
4. The resulting piece has been reset to the default material

Part Design WB has similar problems

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38259 (Git)
Build type: Release
Branch: main
Hash: c8046b1c60c786caca3050cf51f853454ffb5388
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ArchTextures
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * fcgear 1.0.0
  * FeedsAndSpeeds 0.5.0
  * GDML.backup1701064072.5104816 2.0.0 (Disabled)
  * kicadStepUpMod 11.1.2
  * LCInterlocking
  * MaterialTest
  * OpenTheme 2024.5.3
  * render 2024.6.11
  * Rocket 4.0.0
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-18:
Should this issue be fixed for the 1.0 release?

davesrocketshop, 2024-07-19:
Yes

maxwxyz, 2024-08-17:
@davesrocketshop any news on this issue? Is it ok for you if we remove the blocker tag?

davesrocketshop, 2024-08-19:
> @davesrocketshop any news on this issue? Is it ok for you if we remove the blocker tag?

Yes. I'm working on it but a fix may take a while

maxwxyz, 2024-08-20:
so really not necessary for a 1.0 release?

FEA-eng, 2024-08-29:
IMO it should be a blocker for 1.0, unless it's too hard to fix in time. That's a really nasty bug.

davesrocketshop, 2024-08-29:
It is a very difficult fix. I have no clear solution yet, but I'll update if I get close

maxwxyz, 2024-09-13:
Could somebody help with this issue? New duplicate issues coming in regularly now.

davesrocketshop, 2024-09-13:
It is being actively worked on

bgbsww, 2024-09-27:
I'm sure I don't fully understand the nuances here, so feel free to just discard this.

This addresses only the Part Fillets and Chamfers, and it is rough with corner cases, not sure if it's the right direction:

```
App::DocumentObjectExecReturn *Feature::execute()
{
    // belongs in FeatureBase::execute that doesn't exist.
    auto base = dynamic_cast<Part::FilletBase *>(this);
    if ( base ) {
        App::DocumentObject *link = base->Base.getValue();
        if (!link)
            return new App::DocumentObjectExecReturn("No object linked");
        auto mat = Materials::MaterialManager::defaultMaterial();
        auto feature = dynamic_cast<Part::Feature *>(link);
        if (ShapeMaterial.getValue().getName() != feature->ShapeMaterial.getValue().getName())
            if (ShapeMaterial.getValue().getName() == mat->getName())
                ShapeMaterial.setValue(feature->ShapeMaterial.getValue());
    }
    this->Shape.touch();
    return GeoFeature::execute();
}
```

If there's a conversation somewhere I can help with, let me know.

davesrocketshop, 2024-09-27:
> I'm sure I don't fully understand the nuances here, so feel free to just discard this.
>
> This addresses only the Part Fillets and Chamfers, and it is rough with corner cases, not sure if it's the [...]

There are a lot of nuances that I'm trying to track down. This helps.

bgbsww, 2024-09-27:
If there's a section I can hack down some of the weeds in, let me know - I can try to generalize the above and then apply to other part operations if appropriate, or go over and try to play with PD.

maxwxyz, 2024-10-17:
@davesrocketshop is there some state to test or anything other I could help with (not coding unfortunately).

maxwxyz, 2024-10-19:
@davesrocketshop in today's developer meeting it was decided that this should not be a blocker for 1.0
IMO it's a really annoying issue, which is also seen in the multiple duplicate issues which were created. Any chance you could get a fix in 1.0?

davesrocketshop, 2024-10-19:
> @davesrocketshop in today's developer meeting it was decided that this should not be a blocker for 1.0 IMO it's a really annoying issue, which is also seen in the multiple duplicate issues which were created. Any chance you could get a fix in 1.0?

Didn't realize there was a developers meeting today. I should check the calendar more often. I'm almost done with a fix. I hope to have it ready for merge by the end of tomorrow.

davesrocketshop, 2024-10-20:
PR submitted and building #17360

closed 2024-10-21 by yorikvanhavre

JDMils, 2025-08-05:
It's been a while and it seems this is still an issue. I just posted about it on the forums here:
https://forum.freecad.org/viewtopic.php?t=98852

Link to demo video:
https://youtu.be/AoZa4YnJQDE

[Custom_Colours_Lost_JDMils.zip](https://github.com/user-attachments/files/21587448/Custom_Colours_Lost_JDMils.zip)

I'm using one of the latest FC versions:
`OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42780 (Git)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.57/1.5
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * Assembly4 0.50.18
  * Curves 0.6.62
  * fasteners 0.5.38
  * Gridfinity 0.11.9
  * lattice2 1.0.0
  * Manipulator 1.6.0
`

Very simple sample file included.

FEA-eng, 2026-10-31:
Wasn't it at least partially improved some time ago?

JDMils, 2026-11-04:
The main issue seems to be resolved, creating a child object inherits the parent material type.

HOWEVER, it seems we have issues which have stemmed from this fix:
1. The parent object I created has its material set to Copper\Copper-102.
2. If I select a face on the parent object and right-click and select Material, the result is Copper\Copper-102.
3. If I do create a 2D shape on the parent object, such as a circle, and check it's material settings, it is set to Default !?
4. Once I extrude the 2d circle, it's material becomes that of its parent.

- JDMils

PR #17360 (2024-10-21): Materials: Child ignoring parent material
Inherit the material from the parent object when creating a new object, such as during a boolean operation, or when extruding a sketch.

fixes #15503
commit 0804d80e (2024-10-21): Materials: Child ignoring parent material

Inherit the material from the parent object when creating a new object,
such as during a boolean operation, or when extruding a sketch.

fixes #15503
commit c9f0b818 (2024-10-21): Materials: Child ignoring parent material

Inherit the material from the parent object when creating a new object,
such as during a boolean operation, or when extruding a sketch.

fixes #15503