FreeCAD/FreeCAD #15506: Creating symmetry constraint for point causes crash on exit from Sketcher

theosib, 2024-07-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a reliable crash. For this, all you have to do is:
1. Go into Sketch003
2. Exit the sketcher.

It crashes every time.

What I had done was import some external geometry and set a symmetry constraint between two points down near the bottom.
[bracket bracket.FCStd.zip](https://github.com/user-attachments/files/16289742/bracket.bracket.FCStd.zip)
[image]

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-07-18:
Oh, and BTW, something about this crash defeats auto-recovery. Should I file a separate report on that?

theosib, 2024-07-18:
This crash occurs in 38079 also, so I guess the bug has been here a while.

theosib, 2024-07-18:
Also, unlike most crashes I've encountered, the OS actually notices and pops up a Problem Report dialog. Here's what's in the report.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [2415]
Path:                  /Applications/FreeCAD 38231.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-07-18 17:38:35.7045 -0400
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Anonymous UUID:        40F686D5-7298-4D68-A675-DAB8A0D503CE

Sleep/Wake UUID:       9FE26383-6430-4D98-8224-9DFDBCBB02CE

Time Awake Since Boot: 6900000 seconds
Time Since Wake:       1650 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000030
Exception Codes:       0x0000000000000001, 0x0000000000000030

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [2415]

VM Region Info: 0x30 is not in any region.  Bytes before following region: 68719476688
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libFreeCADApp.dylib           	       0x103c984e8 std::__1::__tree_node_base<void*>*& std::__1::__tree<std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::__map_value_compare<Data::MappedName, std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::less<void>, true>, std::__1::allocator<std::__1::__value_type<Data::MappedName, Data::IndexedName>>>::__find_equal<Data::MappedName>(std::__1::__tree_end_node<std::__1::__tree_node_base<void*>*>*&, Data::MappedName const&) + 8
1   libFreeCADApp.dylib           	       0x103c8848c Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) + 832
2   libFreeCADApp.dylib           	       0x103c89adc Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) + 1280
3   Sketcher.so                   	       0x130e289c8 Sketcher::SketchObject::buildShape() + 740
4   Sketcher.so                   	       0x130e213dc Sketcher::SketchObject::execute() + 1040
5   libFreeCADApp.dylib           	       0x103a8977c App::DocumentObject::recompute() + 112
6   Part.so                       	       0x118e4baf0 Part::Feature::recompute() + 20
7   libFreeCADApp.dylib           	       0x103a0cadc App::Document::_recomputeFeature(App::DocumentObject*) + 636
8   libFreeCADApp.dylib           	       0x103a0b624 App::Document::recompute(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*>> const&, bool, bool*, int) + 2284
9   libFreeCADApp.dylib           	       0x103afd7ac App::DocumentPy::recompute(_object*) + 892
10  libFreeCADApp.dylib           	       0x103af6064 App::DocumentPy::staticCallback_recompute(_object*, _object*) + 100
11  libpython3.11.dylib           	       0x101603dec cfunction_call + 284
12  libpython3.11.dylib           	       0x1015ac348 _PyObject_MakeTpCall + 332
13  libpython3.11.dylib           	       0x1016ae680 _PyEval_EvalFrameDefault + 46152
14  libpython3.11.dylib           	       0x1016a2268 PyEval_EvalCode + 220
15  libpython3.11.dylib           	       0x1017083f4 run_mod + 144
16  libpython3.11.dylib           	       0x10170bfa0 PyRun_StringFlags + 112
17  libFreeCADBase.dylib          	       0x10128374c Base::InterpreterSingleton::runString(char const*) + 108
18  libFreeCADGui.dylib           	       0x1026152cc Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 164
19  libFreeCADGui.dylib           	       0x102615128 Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) + 124
20  libFreeCADGui.dylib           	       0x102616660 Gui::Command::updateActive() + 68
21  SketcherGui.so                	       0x137bf1f40 SketcherGui::ViewProviderSketch::unsetEdit(int) + 620
22  libFreeCADGui.dylib           	       0x102984110 Gui::ViewProvider::finishEditing() + 32
[... 1973 lines cut ...]
  ],
  "experiments" : [
    {
      "treatmentId" : "c47ab4cc-c9c3-4b5d-a87c-e2433ce02597",
      "experimentId" : "6639bc6ba73d460582162323",
      "deploymentId" : 400000006
    }
  ]
}
}

Model: Mac14,6, BootROM 8422.141.2, proc 12:8:4 processors, 32 GB, SMC 
Graphics: Apple M2 Max, Apple M2 Max, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Jun 11 2023 05:20:59 version 23.20.95.0.40.50.92 FWID 01-ec505a98
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

xtemp09, 2024-07-19:
@theosib, you have created recently the issue #15102. There is no need to create another issue describing the same problem.

Thowil, 2024-07-19:
I'm not quite sure that this is caused by the symmetry constraint itself.

With weekly build 0.22.0dev.38231 on Win10 64bit I reliably get an access violation message in the report view (not causing freecad to crash though) as soon as a sketch with a single point in it is created.

Start FreeCAD, New File, Open Sketcher and make a new sketch, add a single point, exit sketch -> Access Violation.

Constraints seem to work fine on points as long as those are part of other sketch-features (lines, bezier...).

theosib, 2024-07-19:
> @theosib, you have created recently the issue #15102. There is no need to create another issue describing the same problem.

Are these two the exact same crash?

maxwxyz, 2024-07-19:
Duplicate of #15102

closed 2024-07-19 by maxwxyz