FreeCAD/FreeCAD #15507: Auto-recovery broken in 38231

theosib, 2024-07-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Please see https://github.com/FreeCAD/FreeCAD/issues/15506. I noticed that when it would crash and I restarted FreeCAD, no auto-recovery would happen. One thing I decided to do was see if 38079 suffered from the same crash, but immediately upon loading 38079, the auto-recovery pops up for my project. So this suggests to me that auto-recovery is broken in 38231, and this is a recent regression. 38231 saves backups, but it doesn't try to load them. You can use the project attached to https://github.com/FreeCAD/FreeCAD/issues/15506 to observe this. 

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-21:
can confirm on 
```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41

```

chennes, 2024-08-17:
I could not reproduce with
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

chennes, 2024-08-17:
I also cannot reproduce on
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38475 (Git)
Build type: Release
Branch: pr15929
Hash: 3787fd3b58c50149d778399c54557d817b7a36fc
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```
(which should nearly exactly match the Conda build of the Weekly, save that it's on a newer OS).

chennes, 2024-08-17:
I am going to remove the "RC-Blocker" tag from this one on the basis that it seems to be affecting only a small (possible VERY small, if it's really only affecting macOS 13.6 users). I'll leave it a Blocker for now, but the release candidate should be good to move forward with once all other RC blockers are resolved.

maxwxyz, 2024-09-13:
@chennes without M-series CPU Mac hardware, macOS 13.6 is the latest available version. Would you still consider this a blocker? If not, we could close the entire issue as not planned.

chennes, 2024-09-13:
Oh, I didn't realize that was the sticking point. I still have an Intel laptop in the closet, let me see if I can reproduce there instead.

chennes, 2024-09-19:
Could not reproduce on
```
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38758 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: e19f583e7724c535c6e035d6173f2016ce601f6d
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

chennes, 2024-09-20:
Also cannot reproduce on:
```
OS: macOS 13.7
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38758 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: e19f583e7724c535c6e035d6173f2016ce601f6d
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

chennes, 2024-09-20:
I'm going to drop the "blocker" tag from this issue (but leave the issue open since @maxwxyz can reproduce it).

maxwxyz, 2024-09-20:
I'm fine, unfortunately I'm stuck with this macos version and weird that it seems only to be an issue with 13.6

chennes, 2024-09-20:
I'm surprised that you can't update to 13.7 -- my old Intel Macbook pro was able to (in fact, I could NOT upgrade to 13.6, I was on 13.5 and all they would give me was 13.7).

maxwxyz, 2024-10-02:
I'm closing this if it's no blocker anymore. It certainly won't be fixed for the next release as 13.6 will be even older.

closed 2024-10-02 by maxwxyz