FreeCAD/FreeCAD #15513: Overlay sidebars prevent startpage from functioning.

MisterMakerNL, 2024-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The buttons in the start-page don't work when overlay is over it. 
Result is that the buttons don't work anymore.
Overlay and any sidebar isn't useful when start-page is being used and should be disabled completely in my opinion.
This is a start-page breaking bug and should be fixed before 1.0 in my opinion.
[image]

### Full version info

```shell
dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-07-19:
This is also an issue when using spreadsheets/text editor. Basically anything that isn't the normal 3D view.

Syres916, 2024-07-20:
Something to bear in mind for the developer trying to fix issue this is the plot functionality. I only use it for CfdOF but I imagine FEM uses it also @FEA-eng will no doubt verify. So if a developer uses something like:

```
    if (getActiveGuiDocument())
        return true;
    else
        return false;
```

to determine if a live document is currently being used then this would disable the task panel for plotting see screenshot:

[image]

meaning the user couldn't easily stop it if they could tell from the graph that something was seriously wrong, note some users run solvers only minutes while others leave it overnight say 15 hours.

To assist I ran a simple bit of Python to grab hold of any Qt widgets and the like to find what I believe is the Plot page:

`<PySide2.QtWidgets.QOpenGLWidget(0x557bc8dec430) at 0x7f7c9440da80>, <PySide2.QtWidgets.QMdiSubWindow(0x557bc97c0f30) at 0x7f7c9440dac0>, <PySide2.QtWidgets.QMenu(0x557bcbbcbac0) at 0x7f7c9440db80>, <FreeCAD.Plot.Plot.Plot(0x557bc9500240) at 0x7f7c943ba700>, <matplotlib.backends.backend_qtagg.FigureCanvasQTAgg(0x557bc95002a0) at 0x7f7cb80b4600>, <matplotlib.backends.backend_qt.NavigationToolbar2QT(0x557bc9500300) at 0x7f7cb9400840>`

Hopefully this is of use but if this specific scenario stops what would otherwise be a simple fix then we'll just have to document that non-overlay mode must be used when plotting residuals.

MisterMakerNL, 2024-07-21:
If we disable the overlay mode outside anything but 3D view it would be fixed too.

Syres916, 2024-07-21:
Apologies for my English not being clear, I am asking that the Plot 2D sub window behave opposite to what's proposed for Spreadsheet, Text Editor, StartView. I would like it to be the same as a TechDraw Page tab i.e. task panels must still be accessed even though the live sub window is not a 3D view. Hopefully that clarifies why I posted the above feedback for treating it differently.

MisterMakerNL, 2024-07-22:
Lol you are right, we can just use a setting. :)

PR #15568 (2024-07-22): [Gui] Overlay set hide in non 3D view to be true/checked
Fixes #15513 tested Start module, Macro editor, Spreadsheet, Techdraw page and Plot sub windows which all behave as required.
closed 2024-07-22 by chennes