FreeCAD/FreeCAD #15515: TNP?: Link Copy On Change: Tracking results in many warnings and errors

Roy-043, 2024-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Remove the zip extension and open the attched file: [Table-Body-021.FCStd.zip](https://github.com/user-attachments/files/16309695/Table-Body-021.FCStd.zip).
2. Result: many warnings (in orange). See below.
3. Recompute the file.
4. Result: many messages (in black) and 4 errors (in red). See below.
5. Recompute again.
6. Result: OK.
7. Delete the Fillet.
8. Result: messages and errors from step 4 are again displayed.

None of the messages, warning and errors occur in V0.21.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=89201

Warnings step 2:
```
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Body
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Sketch
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Sketch001
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Pad
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Pad
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Pad001
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Pad001
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#MultiTransform
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Mirrored
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Mirrored001
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Fillet
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Fillet
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Mirrored006
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Mirrored007
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Sketch006
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Pad006
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Pad006
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Sketch007
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Pad007
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Pad007
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#MultiTransform003
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Fillet003
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Fillet003
09:47:10  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: Table_Body_021#Body003
```

Messages step 4:
```
09:48:07  83.8875 <App> Document.cpp(3474): pending remove of Body003 after recomputing document Table_Body_021
09:48:07  83.8876 <App> Document.cpp(3474): pending remove of Origin003 after recomputing document Table_Body_021
09:48:07  83.8876 <App> Document.cpp(3474): pending remove of Z_Axis003 after recomputing document Table_Body_021
09:48:07  83.8877 <App> Document.cpp(3474): pending remove of Y_Axis003 after recomputing document Table_Body_021
09:48:07  83.8877 <App> Document.cpp(3474): pending remove of X_Axis003 after recomputing document Table_Body_021
09:48:07  83.8877 <App> Document.cpp(3474): pending remove of Fillet003 after recomputing document Table_Body_021
09:48:07  83.8878 <App> Document.cpp(3474): pending remove of MultiTransform003 after recomputing document Table_Body_021
09:48:07  83.8878 <App> Document.cpp(3474): pending remove of Pad007 after recomputing document Table_Body_021
09:48:07  83.8878 <App> Document.cpp(3474): pending remove of Sketch007 after recomputing document Table_Body_021
09:48:07  83.8879 <App> Document.cpp(3474): pending remove of Pad006 after recomputing document Table_Body_021
09:48:07  83.8879 <App> Document.cpp(3474): pending remove of Sketch006 after recomputing document Table_Body_021
09:48:07  83.8879 <App> Document.cpp(3474): pending remove of XY_Plane003 after recomputing document Table_Body_021
09:48:07  83.888 <App> Document.cpp(3474): pending remove of Mirrored007 after recomputing document Table_Body_021
09:48:07  83.888 <App> Document.cpp(3474): pending remove of YZ_Plane003 after recomputing document Table_Body_021
09:48:07  83.888 <App> Document.cpp(3474): pending remove of Mirrored006 after recomputing document Table_Body_021
09:48:07  83.888 <App> Document.cpp(3474): pending remove of XZ_Plane003 after recomputing document Table_Body_021
```

Errors step 4:
```
09:48:07  Origin "Table_Body_021#Origin003" doesn't contain feature with role "XZ_Plane"
09:48:07  Origin "Table_Body_021#Origin003" doesn't contain feature with role "XZ_Plane"
09:48:07  Origin "Table_Body_021#Origin003" doesn't contain feature with role "XZ_Plane"
09:48:07  Origin "Table_Body_021#Origin003" doesn't contain feature with role "XZ_Plane"
```

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2024-08-01:
I'm not convinced this is TNP related.  The many warnings seem related to the fixed save/restore bug, but the 4 origin errors don't necessarily seem related.  There is code in LS3 that factors out Origin::setupObject into Origin::initObjects and calls that just before the code in Origin::getOriginFeature in a way that makes me think it would make the errors go away, but I don't directly see the TNP connection so I'm concerned that this might have come from some other set of changes.

bgbsww, 2024-08-01:
Actually, after recalculating and saving the test file without the save/restore problems, it doesn't throw any errors at all when reloaded, recomputed, or deleting the fillet.

I suspect that problems from broken save/restore may have generated everything here.

Roy-043, 2024-08-01:
If I save the file after step 6, reopen it and run the steps again, only the warnings of step 2 are different. All other messages and errors are still there. Maybe you are using a later version?

New warnings step 2:
```
09:06:22  <ElementMap> <string>(1): No hasherRef
09:06:22  <ElementMap> <string>(1): No hasherRef
```

Version:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

bgbsww, 2024-08-09:
There is something funny about the StringHasher table being generated by this file, even after recomputes.  It won't load.  Hopefully I can find that glitch.  I'm suspicious that the MultiTransform is triggering it.

PR #15845 (2024-08-12): Toponaming: Fix transformed; abstract index element name generation
fix #15515  fix #15468 fix #14669 Correct stringhasher save.
closed 2024-08-12 by chennes