FreeCAD/FreeCAD #15518: Availability of tools in the View menu is confusing

Roy-043, 2024-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The availability of tools in the View menu is confusing.

For some tools it seems that the Material WB has to have been loaded (directly or indirectly): Std_SetMaterial and Std_SetAppearance.

For Part_SectionCut the Part or PartDesign WB has to have been loaded.

For Part_ColorPerFace the Part or PartDesign WB must be active.

Two examples:

If the start-up WB is Sketcher and after a restart Part_SectionCut is available. It seems unlikely that the tool is useful for the Sketcher WB.

If the start-up WB is Draft and after a restart Std_SetMaterial and Std_SetAppearance are not available in the View menu. But they are available in the Tree view context menu. Part_ColorPerFace is missing from both menus even though the tool makes sense in the context of the Draft WB.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-12-04:
This issue is still relevant.

@maxwxyz Why is the bot targeting this and other recent issues? And why a bot anyway?

maxwxyz, 2024-12-04:
@Roy-043 we have discussed this at the biweekly dev meeting. To tidy up the backlog. This issue was targeted, as it is not tagged as feature or confirmed bug.