FreeCAD/FreeCAD #15523: BIM: Circular wall broken - index out of range

maxwxyz, 2024-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tired to create a round wall, from a circle (make face = false). This results in an error "Index out of range" in the weekly version. 
There is no issue in the stable 0.21.2 version.

[image]

Steps to reproduce:

1. Create a circle, either from the Draft > Circle tool or create a sketch, Make Face has to be false
2. Select the circle and click the Wall tool
3. Error: Index out of range

When `Make face`= `true` it just extrudes the sketch.
The following demo file by @kaiwas was creates in the stable 0.21.2 version and can be opened in the dev version. As soon as you recompute you get the error:
[Circle_wall.FCStd.zip](https://github.com/user-attachments/files/16313309/Circle_wall.FCStd.zip)

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git)
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-19:
@yorikvanhavre FYI
@bgbsww could this be a TNP issue?

maxwxyz, 2024-07-19:
When I open the circle slightly it works, so maybe it has to do with the closed profile of a circle.

paullee0, 2024-07-20:
@Roy-043, thanks for the fix :)

I have a look at previous implementation and wondering why there was no out of index problem then, Part API change ?  :)
-  https://github.com/FreeCAD/FreeCAD/pull/2746/commits/4a15d8abf32af1b8d479b575f18590e1c2223c2d

Now, remarks in original line #449 #493
                    # if arc/circle
should read ?
                    # if arc

Then, should add remarks to #448 #494 for better understanding ?
                    # if circle

Thanks again :)

Roy-043, 2024-07-21:
IMO these comments are superfluous as they comment what is obvious.

paullee0, 2024-07-22:
Feel free to remove previous remarks altogether :)

PR #15526 (2024-07-29): BIM: Fix index out of range error for circular wall
Fixes #15523.
commit 0338ae3f (2024-07-29): BIM: Fix index out of range error for circular wall (#15526)

* BIM: Fix index out of range error for circular wall

Fixes #15523.

* Reduce comments
closed 2024-07-29 by yorikvanhavre