FreeCAD/FreeCAD #15527: Cannot select all instances of an object in the tree

Roy-043, 2024-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is not possible to select all instances of an object in the Tree view. If there are n instances only n-1 instances can be selected.

1. Remove the zip extension and open the attached file: [select-instances-in-tree.FCStd.zip](https://github.com/user-attachments/files/16315108/select-instances-in-tree.FCStd.zip).
2. Fully expand the tree.
3. Try to select all 4 instances of "Cube".
4. Result: only 3 can be selected.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-19:
Nice bug report but shouldn't the user in this scenario be only able to select _one_ Box/Cube, that was the case on 0.20.x and 0.21.x? Note: I enabled having more than one object with the same label just incase it made a difference but it didn't.

Roy-043, 2024-07-20:
If only one tree instance should be selectable this is a regression.

luzpaz, 2025-06-26:
@Roy-043 any idea what version it was a regression from and perhaps what caused the regression ?

Roy-043, 2025-06-27:
No idea.

Roy-043, 2025-08-27:
Still an issue.
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/20 00:00:55
Build type: Release
Branch: main
Hash: 5e4fe3a3c3af46be530969d323368fee8bfcba3b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

ipatch, 2025-11-20:
fwiw i'm getting the same results with v1.0.2 and a latest dev build as of nov 19 2025, and i'm able to reproduce this issue with both versions.

FEA-eng, 2026-09-30:
Apparently fixed by #26096, can't reproduce it:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

closed 2026-09-30 by maxwxyz