FreeCAD/FreeCAD #15534: In 0.22 opening a sketch breaks object with loft made in 0.21

streb2001, 2024-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Simple object with sketches, pad and two lofts made in FreeCAD 0.21 opened in 0.22. When one of the sketches is opened for editing and then closed, it removes the pad made prior to the loft and inserts an unwanted face where the pad was connected. On the attached project file, double click Sketch001 to edit then immediately close.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

None

### Anything else?

[MPXH6115_pipe_loft1.zip](https://github.com/user-attachments/files/16316547/MPXH6115_pipe_loft1.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-21:
@bgbsww same here I guess

bgbsww, 2024-08-11:
1) Open file.   Lots of Pending Recompute errors, since it doesn't have an element map.
2) Ctrl-r.   4  Modified shae type Wire mismatches.
3) Save file and close.
4) Reopen file
5) No errors, opening and closing Sketch001 does not have any errors or visual differences, don't see a change in the Model.

Is this still an issue?  (There have been a lot of cleanups)

maxwxyz, 2024-08-11:
@streb2001 could you please check in the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
I'll close this then.

closed 2024-08-11 by maxwxyz

streb2001, 2024-08-11:
Seems fine now. Thanks!