FreeCAD/FreeCAD #15538: 0.22: Opening a sketch breaks all external references even without changing anything

knipknap, 2024-07-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The attached document was created in FreeCAD 0.22 and contains two sketches. Opening sketch "Sketch001" breaks all external references. Steps to reproduce:

- Open the document
- Open "Sketch001". Note that everything is well constraint and external references are present
- Close the sketch
- Open it again. Note that all external references are gone.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git) AppImage
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.62.0
  * btl 0.9.9
  * WebTools
  * toSketch
  * fasteners 0.4.62
  * lattice2 1.0.0
  * freecad.gears 1.2.0
  * sheetmetal 0.4.23
  * CycloidGearBox
  * Assembly4 0.12.3
```

### Subproject(s) affected?

Sketcher

### Anything else?

Some logs are shown once the sketch is broken:

```
10:51:05  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: direct_lever_v1#Body
10:51:05  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: direct_lever_v1#Sketch
10:51:05  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: direct_lever_v1#Sketch001
10:51:05  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: direct_lever_v1#Pad
10:51:05  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: direct_lever_v1#Pad
10:51:22  <PropertyLinks> PropertyLinks.cpp(452): direct_lever_v1#Sketch001.ExternalGeometry missing element reference direct_lever_v1#Sketch ;g105;SKT.Edge2
10:51:22  <PropertyLinks> PropertyLinks.cpp(452): direct_lever_v1#Sketch001.ExternalGeometry missing element reference direct_lever_v1#Sketch ;g137;SKT.Edge3
10:51:22  <PropertyLinks> PropertyLinks.cpp(452): direct_lever_v1#Sketch001.ExternalGeometry missing element reference direct_lever_v1#Sketch ;g59;SKT.Edge1
10:51:22  <Exception> TopoShapeExpansion.cpp(2036): Invalid shape name ?Edge2
10:51:22  <PropertyLinks> PropertyLinks.cpp(253): Failed to update element reference of direct_lever_v1#Sketch001.ExternalGeometry
10:51:22  Invalid shape name ?Edge2
Clear constraints to external geometry
10:51:24  direct lever v1#Sketch001: Invalid shape name ?Edge2
10:51:24  direct lever v1#Sketch001: Invalid shape name ?Edge3
10:51:24  direct lever v1#Sketch001: Invalid shape name ?Edge1
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

knipknap, 2024-07-20:
Contrary to what I wrote originally, release 0.22.0dev.37898 is also affected - just slightly differently. While the steps outlined will not break the sketch, the following steps do in that version:

- Open the document
- Open "Sketch001". Note that everything is well constrained and external references are present
- Close the sketch
- Open it again. Note that it still works.
- Open sketch "Sketch", then close it without making a change.
- Open sketch "Sketch001" again. Note that all external references are gone.

maxwxyz, 2024-07-21:
@bgbsww is this also a TNP issue? Should this be closed in favor of another issue?

bgbsww, 2024-07-21:
If it involves reading from an existing file and involves a sketch problem
with external references then it's all the same issue.  All element maps
read back from a 0.22 file are broken right now.

I don't know if they will eventually read cleanly or not.

bgbsww, 2024-08-11:
So, this file clearly has invalid external geo links, most likely used to bugs present in the code at the time it was written.

Unfortunately, it doesn't reconnect on recompute, so I can't intuit where the links were supposed to go.  However, m
any bugs in the code in this area have been corrected, so it would be useful to get confirmation of if this still exists on a new file.  @knipknap 

If it's possible to record a macro while recreating it, then if there's still an issue it will be easier to to track it down.

```
09:34:58  <PropertyLinks> PropertyLinks.cpp(461): direct_lever_v1#Sketch001.ExternalGeometry missing element reference direct_lever_v1#Sketch ;g105;SKT.Edge2
09:34:58  <PropertyLinks> PropertyLinks.cpp(461): direct_lever_v1#Sketch001.ExternalGeometry missing element reference direct_lever_v1#Sketch ;g137;SKT.Edge3
09:34:58  <PropertyLinks> PropertyLinks.cpp(461): direct_lever_v1#Sketch001.ExternalGeometry missing element reference direct_lever_v1#Sketch ;g59;SKT.Edge1
09:34:58  <Sketch> SketchObject.cpp(9177): Failed to project external geometry in direct_lever_v1#Sketch001: Sketch.;g105;SKT
Invalid shape name ?Edge2
09:34:58  <Sketch> SketchObject.cpp(9177): Failed to project external geometry in direct_lever_v1#Sketch001: Sketch.;g137;SKT
Invalid shape name ?Edge3
09:34:58  <Sketch> SketchObject.cpp(9177): Failed to project external geometry in direct_lever_v1#Sketch001: Sketch.;g59;SKT
Invalid shape name ?Edge1
09:34:58  <Sketch> SketchObject.cpp(9259): External geometry direct_lever_v1#Sketch001.e1 missing reference: Sketch.;g105;SKT
09:34:58  <Sketch> SketchObject.cpp(9259): External geometry direct_lever_v1#Sketch001.e2 missing reference: Sketch.;g137;SKT
09:34:58  <Sketch> SketchObject.cpp(9259): External geometry direct_lever_v1#Sketch001.e3 missing reference: Sketch.;g59;SKT
09:34:58  Missing external geometry reference
Clear constraints to external geometry
```

maxwxyz, 2024-08-12:
@knipknap please test with tomorrows weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
And report if this issue is solved or otherwise please provide a macro.

knipknap, 2024-08-14:
I re-created it and this time it doesn't break, so confirming that it works now. Thank you!

closed 2024-08-14 by knipknap