FreeCAD/FreeCAD #15539: Part_RuledSurface fails for two edges with same start point

Roy-043, 2024-07-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Part_RuledSurface fails for two edges with the same start point. The problem does not occur in V0.21.

1. Remove the zip extension and open the attached file: [RuledSurface-Test.FCStd.zip](https://github.com/user-attachments/files/16319766/RuledSurface-Test.FCStd.zip).
2. Recompute the file.
3. Select Line and Line001 (or Line002 and Line003).
4. Invoke Part_RuledSurface.
5. Result: Ruled_Surface: General error in RuledSurface::execute()

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tomate44, 2024-07-28:
Confirmed with : 
```
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git)
Build type: Unknown
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk , OCC 7.8.2.dev
Locale: French/France (fr_FR)
Installed mods: 
  * CurvesWB 0.6.42
```
The bug is triggered at this line : 
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Part/App/TopoShapeExpansion.cpp#L2401
And it seems to happen only when the start point of the first edge is the joining point.

findSubShapesWithSharedVertex() seems quite complex, so I don't know if I will be able to go much further.

drew585, 2024-08-06:
I am experiencing this as well:

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git) AppImage
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods:

[image]
[image]

FC21.3
[image]

maxwxyz, 2024-08-08:
@bgbsww could this be TNP related or a result of different OCC version?

bgbsww, 2024-08-08:
If it is in findSubShapesWithSharedVertex, formerly known as searchSubShapes, and it likely is, then yes, the TNP code changes would be implicated.  Comparison run on LS3 does not show the bug, so this has to be a code transfer error.

First tries at looking at this are hanging the debugger in Geometry::fromShape calling Geometry::makeFromCurveAdaptor on the second edge, as there is no geometry type and it throws a Base::TypeError that is getting swallowed into that debugger hang.   More to come ...

PR #15829 (2024-08-12): Toponaming: Match findSubShapesWithSharedVertex exception suppression
Fix #15539
closed 2024-08-12 by chennes