FreeCAD/FreeCAD #15540: File-dialog appearance problem by PR  #14970

berberic2, 2024-07-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This commit breaks the file-dialog appearance on Linux, KDE X11, introduced by  PR #14970.
The problem is the line
`bool themePaths = hTheme->GetBool("ThemeSearchPaths", false);`
which seems to make the file-dialog not find it’s icons any more und so:
- Toolbar-icons gets replaced by text, taking far too much horizontal space
- The Places Icons are gone or replaced by _wrong_ ones.

Left old, right new.
[image]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38259 +2 (Git)
Build type: Release
Branch: Meins
Hash: cca7ec56c09f1b5e8c9ccd9328066c3277a31d7d
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4 0.50.15
  * sheetmetal 0.4.23
  * Silk 0.1.5
  * CurvedShapes 1.0.9
  * DynamicData 2.62.0
  * FreeCAD_assembly3 0.11.3
  * fcgear 1.0.0
  * fasteners 0.5.24
  * QuickMeasure 2022.10.28
  * ExplodedAssembly
  * Curves 0.6.42
  * Pyramids-and-Polyhedrons
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-20:
@kadet1090 FYI

kadet1090, 2024-07-20:
@maxwxyz can you assign me so it won't get lost  for me?

mnesarco, 2026-01-12:
@maxwxyz @kadet1090 
Is this issue already fixed? I have never experienced this problem on Linux. I see there are some PRs already merged, but it is not clear if this is still an issue.

maxwxyz, 2026-08-15:
Closing for no feedback

closed 2026-08-15 by maxwxyz