FreeCAD/FreeCAD #15542: 0.22DEV

RREE, 2024-07-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The model is a case for the RPi 3A+. Creating sketches for the various holes (SD card, USB-C, USB-A, HDMI, jack) I realized a red exclamation mark in dem model and an "access violation" in the console window. The attached file is a zip of the FCStd file. ([Gehäuse.zip](https://github.com/user-attachments/files/16321831/Gehause.zip))

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Gridfinity 0.5.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-07-21:
Here is the backtrace, indicating that this is a duplicate of #15102. bgbsww [claims that]( https://github.com/FreeCAD/FreeCAD/issues/15102#issuecomment-2241162161) the bug has been fixed in the latest dev build.

@RREE, have you tried to reproduce the bug in the latest weekly bundle?

<details>

```cpp
Process 1750 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x50)
    frame #0: 0x00007ffff6404e77 libFreeCADApp.so`std::map<Data::MappedName, Data::IndexedName, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName>>>::lower_bound(Data::MappedName const&) [inlined] std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName>>, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName>>>::_M_mbegin(this=0x0000000000000040) const at stl_tree.h:737:73
   734 
   735        _Link_type
   736        _M_mbegin() const _GLIBCXX_NOEXCEPT
-> 737        { return static_cast<_Link_type>(this->_M_impl._M_header._M_parent); }
   738 
   739        _Link_type
   740        _M_begin() _GLIBCXX_NOEXCEPT
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x50)
  * frame #0: 0x00007ffff6404e77 libFreeCADApp.so`std::map<Data::MappedName, Data::IndexedName, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName>>>::lower_bound(Data::MappedName const&) [inlined] std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName>>, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName>>>::_M_mbegin(this=0x0000000000000040) const at stl_tree.h:737:73
    frame #1: 0x00007ffff6404e77 libFreeCADApp.so`std::map<Data::MappedName, Data::IndexedName, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName>>>::lower_bound(Data::MappedName const&) [inlined] std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName>>, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName>>>::_M_begin(this=0x0000000000000040) at stl_tree.h:741:25
    frame #2: 0x00007ffff6404e77 libFreeCADApp.so`std::map<Data::MappedName, Data::IndexedName, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName>>>::lower_bound(Data::MappedName const&) [inlined] std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName>>, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName>>>::lower_bound(__k=0x00007fffffffa010, this=0x0000000000000040) at stl_tree.h:1271:30
    frame #3: 0x00007ffff6404e73 libFreeCADApp.so`std::map<Data::MappedName, Data::IndexedName, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName>>>::lower_bound(this=size=0, __x=0x00007fffffffa010) at stl_map.h:1309:32
    frame #4: 0x00007ffff6409df2 libFreeCADApp.so`_ZNSt3mapIN4Data10MappedNameENS0_11IndexedNameESt4lessIvESaISt4pairIKS1_S2_EEE6insertIS5_IS1_S2_EEENSt9enable_ifIXsrSt16is_constructibleIS7_JT_EE5valueES5_ISt17_Rb_tree_iteratorIS7_EbEE4typeEOSE_(this=size=0, __x=0x00007fffffffa010) at stl_map.h:857:31
    frame #5: 0x00007ffff63fb0dc libFreeCADApp.so`Data::ElementMap::addName(this=0x0000000000000000, name=0x00007fffffffa2b0, idx=0x00007fffffffa6c0, sids=0x00007fffffffa298, overwrite=false, existing=0x00007fffffffa2a0) at ElementMap.cpp:507:38
    frame #6: 0x00007ffff63fc8c0 libFreeCADApp.so`Data::ElementMap::setElementName(this=0x0000000000000000, element=0x00007fffffffa6c0, name=0x00007fffffffa710, masterTag=0, sid=0x00007fffffffa298, overwrite=false) at ElementMap.cpp:583:87
    frame #7: 0x00007fffb7e8f810 Sketcher.so`Sketcher::SketchObject::buildShape() [inlined] Data::ComplexGeoData::setElementName(overwrite=false, sid=0x0000000000000000, masterTag=0, name=0x00007fffffffa710, element=0x00007fffffffa6c0, this=0x00007fffffffa8b0) at ComplexGeoData.h:262:86
    frame #8: 0x00007fffb7e8f7dd Sketcher.so`Sketcher::SketchObject::buildShape(this=0x0000555557214200) at SketchObject.cpp:273:34
    frame #9: 0x00007fffb7e901e5 Sketcher.so`Sketcher::SketchObject::execute(this=0x0000555557214200) at SketchObject.cpp:250:15
    frame #10: 0x00007ffff623ee4b libFreeCADApp.so`App::DocumentObject::recompute(this=0x0000555557214200) at DocumentObject.cpp:162:29
    frame #11: 0x00007fffc3470576 Part.so`Part::Feature::recompute(this=<unavailable>) at PartFeature.cpp:105:42
    frame #12: 0x00007ffff61cedcc libFreeCADApp.so`App::Document::_recomputeFeature(this=0x0000555559b4e6c0, Feat=0x0000555557214200) at Document.cpp:3129:41
    frame #13: 0x00007ffff61d7008 libFreeCADApp.so`App::Document::recompute(this=0x0000555559b4e6c0, objs=<unavailable>, force=<unavailable>, hasError=0x0000000000000000, options=<unavailable>) at Document.cpp:2856:48
    frame #14: 0x00007ffff62a4f49 libFreeCADApp.so`App::DocumentPy::recompute(this=0x00005555589a58a0, args=<unavailable>) at DocumentPyImp.cpp:578:54
[... cut ...]
    frame #96: 0x00007ffff3429e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:107:1), argc=1, argv=0x00007fffffffdd48, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd38) at libc-start.c:392:3
    frame #97: 0x0000555555570275 FreeCAD`_start + 37
(lldb)
```

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38239 (Git)
Build type: Debug
Branch: main
Hash: 0278bd8f27df454d34920e0f9a4243783b3e9ebf
Python 3.10.12, Qt 5.15.14, Coin 4.0.0, Vtk , OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

maxwxyz, 2024-07-21:
Duplicate of #15102

closed 2024-07-21 by maxwxyz

xtemp09, 2024-07-21:
In the latest dev build there is neither "Access violation" nor crash. _FreeCAD_ reports:

> Aussparungen vorne: Shape must be a wire, edge or compound. Something else was supplied.

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38275 (Git)
Build type: Debug
Branch: main
Hash: 41d67f8e108a77ec0b98a9ca47d85575eb43476c
Python 3.10.12, Qt 5.15.14, Coin 4.0.0, Vtk , OCC 7.5.1
Locale: English/United States (en_US)
```