FreeCAD/FreeCAD #15543: Pad eats out the whole mode.

GAZ082, 2024-07-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In this case the Pad goes nuts.

https://github.com/user-attachments/assets/804997cc-8f6b-4975-97ff-d2f950bc7d0a

Changed the extention of the file, rename back to .FCStd

[Cuelga Toalla V2.zip](https://github.com/user-attachments/files/16322300/Cuelga.Toalla.V2.zip)

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git) AppImage
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves.backup1713713759.0032098 0.6.29 (Disabled)
  * QuickMeasure 2022.10.28
  * ExtremeProDark 2.9.0
  * Curves.backup1693059780.2922277 0.6.13 (Disabled)
  * CfdOF.backup1710694906.6805208 1.24.9 (Disabled)
  * Assembly4.backup1696800837.713216 0.50.2 (Disabled)
  * Curves.backup1714093860.7275996 0.6.34 (Disabled)
  * OpenTheme 2024.5.16
  * CfdOF.backup1699129791.898633 1.24.7 (Disabled)
  * Assembly4.backup1710902627.637761 0.50.9 (Disabled)
  * Defeaturing.backup1710696629.0288966 1.2.2 (Disabled)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-20:
Opened your file, deleted _Pad003_, marked the whole document for Recompute, Refresh, start _Pad_ operation of _Sketch014_ and click _Reversed_ now successful Pad feature using:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38265 (Git)
Build type: Release
Branch: main
Hash: 608cb182e516e683befa890a84ca615c040b1235
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * CfdOF 1.27.4
```

GAZ082, 2024-07-20:
Odd, opened my own file, recalculated (as I usually do) and now works... *sigh* Dont know if close this or not, because this happened for real.

Syres916, 2024-07-20:
Well we are still in the midst of code for the Topo Naming being implemented so for me it's not worth keeping open because in a week's time the behaviour could well have changed, it's not like there's no-one testing/testing/testing when these changes are being merged. I'm trying to increase the stress level of the tests so we can capture as much as possible but it's not an overnight piece of work.

closed 2024-07-21 by GAZ082