FreeCAD/FreeCAD #15545: [ Sketch Geometry Extension ] - Not Working on Point Object

paullee0, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The problem was first discussed on FC forum ON 23.5.2024 [[Bug?] - Sketch Geometry Extension Not Saved in Point Object](https://forum.freecad.org/viewtopic.php?t=86350)  :-
. (cross-posting below)
.
.
.
.
" I understood @ abdullah (on FC forum) developed the Sketcher Geometry Extension - [Sketcher Development - Integration of Extensions](https://forum.freecad.org/viewtopic.php?t=51716)

It seems there is a bug here? Not working fully with Point object? :)

**Demo model file :**

(Forum Link - [Test_ Sketch_ 001_ Point-SetExtension.FCStd](https://forum.freecad.org/download/file.php?id=258487) )
[Test_ Sketch_ 001_ Point-SetExtension.FCStd.zip](https://github.com/user-attachments/files/16323152/Test_.Sketch_.001_.Point-SetExtension.FCStd.zip)

**How to Reproduce the Bug -**

1.  A Sketch contains randomly 15 geometries, 3 are Points

[image]

2.  Select it and run below script, so every geometry (15) is set with GeometryStringExtension 'uuid'

```
    import uuid
    sk=Gui.Selection.getSelection()[0]
    sg = sk.Geometry
    for i in sg:
        us = None
        if i.hasExtensionOfName('uuid'):
            print(' Geometry uuid str is present')
        else:
            print(' Geometry has no Extension "uuid" ')
            u = uuid.uuid4()
            us = str(u)
            i.setExtension(Part.GeometryStringExtension(us,"uuid"))
            print(' Set Geometry Extension "uuid" now')
```

3.  The above return 15 pair of messages ' Geometry has no Extension "uuid"' & ' Set Geometry Extension "uuid" now'

```
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now
     Geometry has no Extension "uuid" 
     Set Geometry Extension "uuid" now

```    
4.  Run below to verify all geometry has GeometryStringExtension 'uuid' now

```
    for i in sg:
        if i.hasExtensionOfName('uuid'):
            print(' Geometry uuid str is really present')
        else:
            print(' Geometry has no Extension "uuid" ')

```    

5.  It should return all GeometryStringExtension are set

```
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
```
     
6.  Now, copy back the geometries with the Extensions to the Sketch Object and test the Sketch.Geometry again

```
    sk.Geometry = sg  # Copy back to Sketch object
    sgNew = sk.Geometry
    for i in sgNew:
        if i.hasExtensionOfName('uuid'):
            print(' Geometry uuid str is really present')
        else:
            print(' Geometry has no Extension "uuid" ')
```

7.  Only the (3) Point objects do not have the GeometryStringExtension 'uuid' written back

```
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry has no Extension "uuid" 
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry uuid str is really present
     Geometry has no Extension "uuid" 
     Geometry has no Extension "uuid" 
```

Anything missing ?

(I use Sketcher::SketchObjectPython, and same result)

### Full version info

```shell
OS: Fedora Linux 38 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ArchTextures
  * Reinforcement
  * Link to SketchArchPlus
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

External SketchArch add-on

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

paullee0, 2024-12-05:
OK, tested with latest git AppImage, on Fedora 40, same result :)

[image]

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Link to SketchArchPlus

```

paullee0, 2025-09-17:
@FEA-eng   Any usecases for reference?  Thanks

FEA-eng, 2025-09-17:
> [@FEA-eng](https://github.com/FEA-eng) Any usecases for reference? Thanks

Unfortunately not, I only tested it with your example to see if it's still there and prevent it from being automatically closed as inactive.

closed 2026-09-08 by maxwxyz

paullee0, 2026-09-12:
Tested with below AppImage version, thanks, it should works now :D

[image]

OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 26.3.0dev.20260908 (Git shallow) AppImage
Build date: 2026/09/08 23:17:32
Build type: Release
Branch: (HEAD detached at 306259d7a)
Hash: 306259d7aa3e7f3b40a22232eb60973deb39ca20
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/30.48/1
OpenGL version/vendor/renderer: 3.3 (Compatibility Profile) Mesa 24.1.7/Intel/Mesa Intel(R) HD Graphics 2000 (SNB GT1)