FreeCAD/FreeCAD #15546: Offset icon is missing

3DDuck, 2024-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Missinf feature "Offset" in sketcher
[image]

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (X-QUBES/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.23
  * Glass
  * Render 2024.7.4
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

3DDuck, 2024-07-21:
> ### Is there an existing issue for this?
> 
>     * [x]  I have searched the existing issues
> 
> 
> ### Problem description
> 
> Missing feature "Offset" in sketcher ![Screenshot_2024-07-21_03-33-00](https://private-user-images.githubusercontent.com/167621231/350744534-f51ae7cb-f26e-4feb-b46f-c0687bfd16c3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1NDc1NjMsIm5iZiI6MTcyMTU0NzI2MywicGF0aCI6Ii8xNjc2MjEyMzEvMzUwNzQ0NTM0LWY1MWFlN2NiLWYyNmUtNGZlYi1iNDZmLWMwNjg3YmZkMTZjMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyMVQwNzM0MjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNTQ5Zjk0MjZlYjJjMGE1Y2MyNzJkZTcxMDQxMjExNTE2YzlkNDUyZWM5ZjQwZGE3OTRjOGI5MTQzM2E2ZDIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.NmIiagNBQgSdcbX1HgN5c_l3H_Gjm_8pN6YsW_EN7yE)
> ### Full version info
> 
> ```shell
> [code]
> OS: Debian GNU/Linux 12 (bookworm) (X-QUBES/xfce)
> Word size of FreeCAD: 64-bit
> Version: 0.20.2.
> Build type: Release
> Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
> Locale: English/United States (en_US)
> Installed mods: 
>   * sheetmetal 0.4.23
>   * Glass
>   * Render 2024.7.4
> [/code]
> ```
> 
> ### Subproject(s) affected?
> 
> None
> ### Anything else?
> 
> _No response_
> ### Code of Conduct
> 
>     * [x]  I agree to follow this project's Code of Conduct

FEA-eng, 2024-07-21:
It was added in 0.22 dev version, you won’t find it in 0.20.2.

closed 2024-07-21 by maxwxyz